grandideastudio/jtagulator #3: Reporting incorrect / inconsistent IDCODE from BeagleBone Black

BobHeinemann, 2015-11-05:
I've encountered an inconsistency when decoding the IDCODE for the BeagleBone Black. I've tested this on both JTAGulator firmware versions 1.1.1 and 1.2.2.

 The JTAGulator correctly identifies the JTAG pins (TDI, TDO, TMS).  However when I perform the "Get Device ID(s)" function the following IDCODE is returned:

Device ID: 1111 1110000110000000 00000000111 1 (0xFE18000F)
-> Manufacturer ID: 0x007
-> Part Number: 0xE180
-> Version: 0xF

This IDCODE is different then what openOCD return when performing an autoprobe.  This is what openOCD returns:

Open On-Chip Debugger 0.7.0 (2014-03-06-11:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.sourceforge.net/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
adapter speed: 500 kHz
Info : clock speed 500 kHz
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -expected-id 0x2b94402f ..."
Warn : AUTO auto0.tap - use "... -irlen 6"
Warn : gdb services need one or more targets defined

Two different IDCODES for the same board.

JTAGulator Reports: 0xFE18000F
openOCD Reports: 0x2b94402f 

The openOCD IDCODE is consistent for the am335x processor which is on the BBB.  The IDCODE that JTAGulator reported, I could find not reference for.

Although I am not certain I am leaning to the openOCD IDCODE to be the correct one.

joegrand, 2015-11-06:
What you're seeing is a known bug and I've been able to recreate the issue with a few different targets (BBB, Pogoplug, and others). My two initial thoughts are it may be a timing issue affecting certain targets or possibly a hardware issue where the JTAGulator is preventing the target from returning the proper data. I need to carve out some time to identify the root cause - this is high on my list.

BobHeinemann, 2015-11-07:
Ah ok.  Was that documented somewhere? I might s ok end some time looking
at it with a LA to see if the BBB is at least returning the right idcode.

I also encountered another issue with the BBB.  It had to do we it TCK and
RTCK  being tied together.  It causes the IDCODE scan to return two
separate results.  The BYPASS scan returns many different false positives.
Have you seen that before?

Bob

joegrand, 2015-11-07:
It is now :) https://github.com/grandideastudio/jtagulator/blob/master/TODO.markdown  It could be a few weeks (at least) before I have the time to look at this issue in full detail. 

Some preliminary captures I took a while back with a Pogoplug showed impulses instead of full square waves on TDO. The high impulses corresponded to actual, correct IDCODE, but weren't interpreted properly by the JTAGulator. I'm not sure if the JTAGulator hardware is causing the impulse or if it's a timing issue or perhaps state of the Propeller I/O pins at that point in time.

[image]

I haven't tried any devices with TCK and RTCK tied together. There isn't support for RTCK yet and I'm not sure how a target will respond. You may want to open this as a separate issue.

joegrand, 2015-11-07:
It would be interesting to know if you're seeing something similar on the BBB.

joegrand, 2015-11-20:
I believe I've identified the root cause of the impulses, which in turn are resulting in incorrect IDCODE reads. It has to do with the front end (level shifting/current limiting) and am putting together a test jig to verify that.

Regarding the BYPASS scan resulting in many positives, I just modified the code today (and pushed to github) to perform additional sanity checks during the scan: 1) Limit the maximum number of devices in the chain to 32, 2) Actually perform a BYPASS test on the detected pinout. The BYPASS scan just shifts a bunch of ones into the IR and then counts how long a zero will take to traverse the chain. So, it could see many false positives if a 0 just happens to be on the pin the JTAGulator is currently looking at as TDI.

BobHeinemann, 2015-11-21:
When I get a chance to test the code I'll let you know how it goes.

Ditto on taking some signal captures on the BBB.

ssantner, 2016-09-07:
Regarding the incorrect IDCODE reads, I'm seeing the same issue on the BBB using the latest stable release of the firmware (v1.3).

You had mentioned previously that the root cause was related to the front end (level shifting and current limiting) - have you been able to verify that?  Is there a workaround that you have identified?

joegrand, 2016-09-08:
I haven't made much progress on this issue, but hope to be able to focus more on it (and other JTAGulator updates) later in the year. I didn't have any luck with my initial attempt with a modified front end using lower value current limiting resistors and pulldowns, so it will take some more time to really dig down and identify the root cause of why certain targets are behaving strangely. Sorry!

joegrand, 2017-04-15:
This is still in-progress and not an easy task. I have a handful of target boards with similar problems. I'm expecting that solving the issue (which may be a combination of firmware and hardware front end) will increase the overall reliability of the tool.

stock99, 2017-10-09:
run into similar issue here with the router (n3g002w http://support.netcommwireless.com/sites/default/files/N3G002WSpecSheet.pdf ).  The idcode scan doesn't appear to give same result.  When i entered some of the result into bypass scan, i get 'no target device found' message. 
[jtagulator_output.txt](https://github.com/grandideastudio/jtagulator/files/1369882/jtagulator_output.txt)
[image]
[image]
[image]

the strange part is that even the uart scan also fail with the same error message. Pretty sure it is not related to my unprofessional solder as i got the same result when using clip probe cables(prior soldering the headers). 

ps: i am hardware noob too. Pardon me if this isn't a related issue.

joegrand, 2018-05-31:
I've written some code that I hope will increase the reliability on certain targets (related to the JTAG clock and when I'm sampling incoming data). Hoping to test this soon and I'll report back when I do.

joegrand, 2020-06-18:
Sorry for such a long delay without an update! I'm really hoping to nail this down once and for all.

As I've previously mentioned, I believe the incorrect IDCODE results or lack of detection on certain targets are due to the JTAGulator's front-end hardware (e.g., TI TXS0108E level translators and/or series resistors used for input protection and/or signal bouncing/reflections). 

I ran into this recently when integrating support for ARM SWD detection (#30), where the target pull resistors were causing the level translators to not convert the voltages properly, so either the Propeller or target weren't receiving the proper logic signals. Bypassing the front-end circuitry let us finish the SWD development. I've seen that with some JTAG hardware, as well, particularly with strong pull resistors that form too large of a voltage divider with the JTAGulator's series resistors (which are used for input protection). 

There may be multiple issues at play, but I think they all fall on the front-end hardware and not on the JTAG software implementation. 

For example, using a LoFive R1.1 module for testing, I get the following results:

With a stock JTAGulator:
```
Device ID #1: 1111 0000000000000000 00000000001 1 (0xF0000003)
-> Manufacturer ID: 0x001
-> Part Number: 0x0000
-> Version: 0xF
```
This is a correct detection of the JTAG pins, but an incorrect reading of the IDCODE. 

With a Propeller Professional Development Board (which is essentially a JTAGulator without the front-end circuitry and target voltage generation):
```
Device ID #1: 0010 0000000000000000 10010001001 1 (0x20000913)
-> Manufacturer ID: 0x489
-> Part Number: 0x0000
-> Version: 0x2
```
This is a correct detection of the JTAG pins *and* a correct reading of the IDCODE. 

Using the Propeller Professional Development Board with a Lattice LCMXO256C FPGA development board usually gives the correct IDCODE:
```
Device ID #1: 0000 0001001010000001 00000100001 1 (0x01281043)
-> Manufacturer ID: 0x021
-> Part Number: 0x1281
-> Version: 0x0
```
But, once in a while there's an impulse on TDO instead of the full pulse, which results in an incorrect IDCODE:
```
Device ID #1: 1111 1000000000000000 00000000001 1 (0xF8000003)
-> Manufacturer ID: 0x001
-> Part Number: 0x8000
-> Version: 0xF
```
```
Device ID #1: 1110 0000001001000000 00000000001 1 (0xE0240003)
-> Manufacturer ID: 0x001
-> Part Number: 0x0240
-> Version: 0xE
```
This could be due to noise or reflections/bouncing on the clock line and could maybe be squashed with a small series resistor. The board never returns the correct IDCODE using a stock JTAGulator. 

Using the Propeller Professional Development Board with a BeagleBone Black (the original subject of this GitHub issue), I get the correct detection of JTAG pins and a correct reading of the IDCODE:
```
Device ID #1: 0010 1011100101000100 00000010111 1 (0x2B94402F)
-> Manufacturer ID: 0x017
-> Part Number: 0xB944
-> Version: 0x2
```
Curiously, I *also* get a correct reading with a stock JTAGulator, even with the original firmware v1.0 from 2013!

Anyway, my plan is to continue experimenting with component changes of the front-end circuitry to see if that helps reduce failures. If hardware changes are required (which it looks like they will be), then I'd create Rev. C hardware. Hopefully, the changes would be minimal enough for people to do on their own Rev. B hardware, but that remains to be seen.

joegrand, 2020-06-18:
@adamgreen [posted this](https://github.com/grandideastudio/jtagulator/pull/30#issuecomment-645620593) regarding the TXS0108 that may be relevant here:

>> I was doing some searching on the inter webs a week or so ago for information about the TI TXS0108 and SWD/JTAG when I came across this discussion on the [Black Magic Probe's Gitter feed](https://gitter.im/blacksphere/blackmagic?at=5b29ac8cad21887018c92735):

>Piotr Esden-Tempski @esden Jun 13 2018 12:21
@hei5enbrg the main reason why we went away from the TXS is that it has really weak drive power. Due to the "magical" and "automatic" direction control it fights basically itself all the time. If you are usig a cable that is not very short has ideal quality and there are no resistors or other protection of the target circuitry it is not working well. When we figured that this is an issue some users have I designed the version with dedicated direction switchable dual supply buffers. This made the issues go away.
@hei5enbrg so if you are asking if you should use the TXS in some project of yours. Unless you control both sides of the connection all the time do NOT use it. You and your customers will be unhappy.

A challenge with the JTAGulator is that we can't control what signals or hardware will be connected to what channels. It may be that any auto-direction detecting level translator will be problematic for us, since targets will have varying pull resistor values that could conflict with the internal resistors in the translators. A suitable translator may exist that is more "friendly" to varying external resistors, but I'm not sure about that yet. In most situations, we seem to be fine using the TXS0108E, but certainly not for the SWD implementation and it could be part of our issue here. Regardless, we'll need both level translation *and* bidirectional communication per pin. That doubles the fun. More experiments are required!

joegrand, 2020-06-25:
From [RECESSIM_ on Twitter](https://twitter.com/RECESSIM_/status/1275775651983826944):

> Recently purchased a JTAGulator and noticed on two boards it wouldn’t discover JTAG, even though on one board I was 100% sure I was connected. TRST line being pulled low by onboard 2.2k resistor seems to be an issue. Documented at https://wiki.recessim.com/view/JTAGulator

This is a related issue which resulted in no detection of the target (as opposed to proper detection but incorrect identification). In this case, the JTAGulator's series resistors (1k) used for input protection are causing too large of a voltage divider when combined with the target's pull resistors. The target has 2.2k resistor values, which are quite strong for JTAG (compared to typical values of 4.7k-10k, which shouldn't cause the voltage level issues). This is something that will be addressed with the front-end redesign.

vladionescu, 2021-01-04:
@joegrand will Rev B boards be customizable/bodge-able to become Rev C? I wouldn't mind hacking up my Rev B to get reliable SWD support :smile:

joegrand, 2021-01-05:
That's the plan. I'm currently working on a [Hardware Modifications](https://github.com/grandideastudio/jtagulator/wiki/Hardware-Modifications) entry for the [Wiki](https://github.com/grandideastudio/jtagulator/wiki) that details the recommended fixes to enhance target compatibility. The trade-off being that you likely *won't* be able to maintain input protection. I'm still investigating the best plan of attack with regards to modifications and a possible hardware revision. I'll report back when done!

vladionescu, 2021-01-05:
Thanks Joe for your awesome support 😃 !

blurpy, 2021-07-05:
I see the wiki page for hardware modifications is linked in different issues, but it doesn't contain any information yet. Is it supposed to?

Also very curious if there's any update on a new revision :)

joegrand, 2021-07-05:
I'm still working on the [Hardware Modifications](https://github.com/grandideastudio/jtagulator/wiki/Hardware-Modifications) document and need to test a few more options. I don't have a timeframe on it, but it might be done by the end of summer.

I've decided to not release a new revision of hardware.

nemanjan00, 2022-06-10:
@joegrand, any updates on this?

I am getting wrong device ID connecting to Rapberry pi 4

ojg78, 2022-11-26:
Joe, is the hardware mods document still planned?

joegrand, 2022-11-26:
Yes. I have a draft ready to go and it will be posted shortly! Just need to verify some of the component-specific details...

dadapa5, 2023-02-20:
When will the new version be released

joegrand, 2023-03-06:
It took years longer than expected, but the [Hardware Modifications](https://github.com/grandideastudio/jtagulator/wiki/Hardware-Modifications) document is finally done! Thanks for your patience :)

The document describes two modifications that you can do to your JTAGulator HW Rev. B to improve target compatibility at the expense of reduced input protection and/or level translation. I don't have plans to create a HW Rev. C that incorporate these changes.

The most offending component is the 1 kΩ series protection resistor on each channel used in the JTAGulator's front end. This is overkill for most situations and causes significant resistor/voltage divider issues with targets that have strong (< 4.7 kΩ) pull resistors. This modification is the simplest of the two and should be effective in most situations. For extreme cases, you can also perform the second modification that completely removes the level translators and replaces them with additional series resistors. Of course, in that case you lose the adjustable target voltage capability, so you're limited to targets that have 3.3 V signals. I explored some options in replacing the level translators with other types, but couldn't find a suitable solution that didn't have its own set of problems.

I've tested this on all the targets in this thread and #30 that have previously had issues.

I'll leave this issue open for now to increase visibility.

joegrand, 2024-03-26:
I think it's safe to close this issue now. The [Hardware Modifications](https://github.com/grandideastudio/jtagulator/wiki/Hardware-Modifications) document has been up for over a year and is also mentioned in the [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions).

closed 2024-03-26 by joegrand