grandideastudio/jtagulator #6: Multiple JTAGS

senadaruc, 2016-01-28:
Dear.

I am trying to identify a JTAG on a device where I receive different values every time!

Do i need to put the device ON or OFF ?

Thanks,

joegrand, 2016-01-28:
The target device should be turned on while you are JTAGulating. Be sure that GND is properly connected between the JTAGulator and target, and that VADJ is NOT connected to the target. If that doesn't solve the problem, then it could be related to issue #3. Documentation and videos are available from http://www.grandideastudio.com/portfolio/jtagulator/.

senadaruc, 2016-01-29:
Dear Joe,

Thanks for the prompt response and this is the output when I run the "I"

> Enter starting channel [0]:
> Enter ending channel [8]:
> Possible permutations: 504
> Press spacebar to begin (any other key to abort)...
> JTAGulating! Press any key to abort...
> ----
> TDI: N/A
> TDO: 8
> TCK: 2
> TMS: 0
> TRST#: 1
> TRST#: 3
> TRST#: 4
> TRST#: 5
> TRST#: 6
> TRST#: 7
> -
> IDCODE scan complete.
> 

this is for the command "D"

> TDI not needed to retrieve Device ID.
> Enter TDO pin [8]:
> Enter TCK pin [2]:
> Enter TMS pin [0]:
> All other channels set to output HIGH.
> No target device(s) found!
> IDCODE listing complete.
> :

and this is for the command "B"

> :B
> Enter starting channel [0]:
> Enter ending channel [8]:
> Are any pins already known? [y/N]: n
> Possible permutations: 3024
> Press spacebar to begin (any other key to abort)...
> JTAGulating! Press any key to abort...
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------
> ----------------------------------------------------------------------------------------------------------------------------------------------
> No target device(s) found!
> BYPASS scan complete.
> :+1: 

The device is ON and Jtagulator works good.

Any help on this.

joegrand, 2016-01-29:
Can you try the T command to test if BYPASS works properly? Your original post says "Different data every time." Is this the case? Also, please attach pictures of your setup and details of the target (vendor, model number, etc.)

closed 2016-06-01 by joegrand