grandideastudio/jtagulator #7: Problems finding the JTAG port using the jtagulator

hyojinjo, 2016-08-29:
Hello Joe, 
I'm struggling to find the jtag port using the jtagulator
I bought the jtagulator a month ago, and I wanted to check that this can find the jtag port corretly.
So I tested it using a board whose jtag port is known to the public. 
The board name is MSP-EXP430F5438 produced by the Texas Instrument and it has the jtag port as below picture. 
[image]

However, I could not find the jtag. The jtagulator did not give answers at all as shown in belows.
[image]
[image]

Even if i tested the "t command" using the TDI and TDO info ,which are from the public document, the jtagulator gave me only "nomatch" results.
[image]

The test envrionment is like that.
[image]
[image]

I don't know what is the problem. I just follows the instructions of your YouTube vedio and set the correct voltage and did not connect VADJ to the board. 
Does my jtagulator have the problems?  
Could you give me helpful advices?

Thanks.

hyojinjo, 2016-08-29:
When I performed the test, the board was powered on.

Thanks,

joegrand, 2016-08-30:
Hi Hyojinjo-

It's possible that you need to set RST/NMI to the correct state. I don't know if that is an active HIGH or LOW reset, so you should check the data sheet to make sure what's needed for normal system functionality. I can't tell from the pictures if you have that pin connected to the JTAGulator - if you did, it would be held HIGH which may be holding the target in reset.

If that doesn't work, can you let me know what pins are connected to what colored wires?

Joe

hyojinjo, 2016-09-01:
Hi Joe,

I checked the data sheet in the link 
(http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=slau263&fileType=pdf)
I think that RST/NMI is set HIGH. Plesae look the below picture.
[image]

RST/NMI is connected to the S3 switch on the board. 
So, I performed the following 2 tests for jtagulating the board
 1. Press and hold the S3 switch while jtagulating   
 2. Do not press S3 switch

However, 2 tests could not find JTAG pins at all. 
My experiment envionment as shown in below.
[image]
[image]
[image]
[image]

Thank you for your help in advance.

joegrand, 2016-09-02:
Hi Hyojinjo-

Note that the /RST pin on the JTAG connector is NOT the JTAG /TRST pin (which is optional per the JTAG spec). Your /RST is an active low system reset. By holding the S3 switch, you are effectively holding the MSP430 in reset, which would prevent JTAG from working.

Upon further review of the MSP430 data sheet, it looks like for some lower pin count MSP430 devices, the JTAG interface is shared with GPIO pins and default to the GPIO (General Purpose Input/Output) state. In order to enable JTAG functionality, the TEST pin must be pulled HIGH (this pin is normally held low with an internal pull-down resistor). Please see the following portions:

MSP430F543xA, MSP430F541xA Mixed-Signal Microcontrollers Data Sheet
http://www.ti.com/lit/ds/slas655e/slas655e.pdf#page46
"The MSP430 family supports the standard JTAG interface which requires four signals for sending and receiving data. The JTAG signals are shared with general-purpose I/O. The TEST/SBWTCK pin is used to enable the JTAG signals."

MSP430 Programming With the JTAG Interface
http://www.ti.com/lit/ug/slau320x/slau320x.pdf#page8
"The standard JTAG interface requires four signals for sending and receiving data. On larger MSP430 devices, these pins are dedicated for JTAG. Smaller devices with fewer total pins multiplex these JTAG lines with general-purpose functions. On these smaller devices, one additional signal is required that is used to define the state of the shared pins. This signal is applied to the TEST pin."

The TEST signal is available on your JTAG connector at pin 8. From what I can tell, you do not have this pin connected to the JTAGulator (which would pull it HIGH while it's not being used as part of the pin enumeration) or otherwise pulled HIGH. Can you please try that? To simplify the experiment, you can disconnect /RST. It shouldn't make a difference, though.

Joe

hyojinjo, 2016-09-06:
Thank you for your comments.
I will find how to enable the JTAG.

joegrand, 2016-09-08:
Hi Hyojinjo-

Another thing you can try is to simply hook up JTAGulator channels to all of the pins on the known JTAG connector. This is really the best/intended use of the JTAGulator, as it will enumerate through the possible pin permutations and you don't have to worry about dealing with individual pins or their functions. Any pin that isn't currently being tested will be set to HIGH, which is helpful if you are connected to an active low JTAG reset (/TRST) or system reset pin (/SRST). Or, in the case of the MSP430, the TEST pin that enables JTAG. Problems arise, as you've had, when not all of the required pins are connected to the JTAGulator. 

Joe

closed 2016-10-12 by joegrand