grandideastudio/jtagulator #10: higher channel port numbers less reliable ?

cederom, 2017-04-15:
Hey Joe =) I did some self assembed JTAGulator. I find that higher ports tends to be less reliable, sometimes they dont reply valid pinout at the first time but after second or third they work, also making some cable permutation of the same signal sometimes helps. I did not find that problem with lower ports (i.e. CH0..CH7). Any ideas what could be the reason? All signals goes well on the oscilloscope..

joegrand, 2017-04-15:
Since you're assembling your own hardware, there's not much I can do to help you troubleshoot. The official JTAGulator units sold through Parallax and their authorized distributors are fully tested during the manufacturing process and we haven't seen what you're describing. You can try to follow the test procedure available at http://www.grandideastudio.com/jtagulator/ to ensure that all of your channels are working properly without any short circuits.

closed 2017-04-15 by joegrand

cederom, 2017-04-16:
Thank you Joe, I noticed that only on one of five I did, probably a component fault, that was cheaper and more fun to DIY, if I sell any will share some ca$h no worries, happy easter! :-)

K1two2, 2020-04-26:
> 由于您正在组装自己的硬件，因此我无法帮助您排除故障。通过视差销售的官方 JTAGulator 装置及其授权分销商在制造过程中经过全面测试，我们尚未看到您描述的内容。您可以尝试遵循http://www.grandideastudio.com/jtagulator/可用的测试程序，以确保所有通道都正常工作，而不会有任何短路。

English is not my mother tongue, can't understand your handwritten English font, can you send a TXT text?