grandideastudio/jtagulator #11: VADJ no use?

cederom, 2017-04-15:
Hey Joe =) What is the purpose of VADJ input when I have to select VCC for JTAG port manually and VADJ needs to be disconnected? Some future use?

joegrand, 2017-04-15:
VADJ is the output from the PWM/op amp hardware on the JTAGulator that's used to create the target system voltage and drive the on-board level translators. It's nice to have accessible on the channel pins for testing to make sure the target system voltage is being generated corrected and thought it might be nice to have for future use (but don't have any particular ideas in mind). If you connect VADJ to a target, it could load down the voltage output and affect the results from the JTAGulator.

closed 2017-04-15 by joegrand

cederom, 2017-04-16:
ah so its the output not the input.. on KT-LINK multi-purpose interface there is a VADJ input so you can connect to target VCC and voltage shifters will do the job with setting VCC :-)