grandideastudio/jtagulator #9: TDI not identified by IDCODE scan but found by BYPASS scan

cederom, 2017-04-15:
Hey Joe! =)

I found that IDCODE scan does not always find TDI. It gives N/A but correctly finds other signals. BYPASS scan however finds TDI with no problem, and I find BYPASS scan more reliable, even on SECURE JTAG ports.. FW 4.0.

Best regards :-)
Tomek

joegrand, 2017-04-15:
The IDCODE Scan does not look for TDI, it only looks for TDO, TMS, TCK. The reason is that most devices that contain an IDCODE (which is optional per the JTAG specification) will place the IDCODE into the DR on power-up. So, all we have to do is shift into the Shift-DR state (using TMS and TCK) and then shift whatever data is in the DR via the TDO pin. The IDCODE Scan will try all permutations of channels looking for what may be a valid IDCODE output on TDO. Once you find the three lines, you can then use the BYPASS Scan to find TDI. I usually use the IDCODE Scan first against an unknown target, since it's faster and will at least give me a preliminary result as to if there's JTAG functionality available. You can find more details about that command in my video here: https://youtu.be/GgMOBhmEJXA?t=13m50s

closed 2017-04-15 by joegrand

cederom, 2017-04-16:
nice trick! tanks for hint! =)