grandideastudio/jtagulator #13: cJTAG support

hei5enberg, 2017-05-25:
Hi Joe,

I've tried searching everywhere. Please clarify.

I'm working with an arm cortex-m4 micro-controller ([datasheet](http://www.st.com/content/ccc/resource/technical/document/datasheet/b3/a5/46/3b/b4/e5/4c/85/DM00115249.pdf/files/DM00115249.pdf/jcr:content/translations/en.DM00115249.pdf)) which has a SDW interface and a 2 pin JTAG (I guess it is cJTAG / IEEE 1149.7) interface. I already know that jtagulator does not support SWD.

from the datasheet
```
The ARM SWJ-DP interface is embedded, and is a combined JTAG and serial wire debug
port that enables either a serial wire debug or a JTAG probe to be connected to the target.
Debug is performed using 2 pins only instead of 5 required by the JTAG (JTAG pins could
be re-use as GPIO with alternate function): the JTAG TMS and TCK pins are shared with
SWDIO and SWCLK, respectively, and a specific sequence on the TMS pin is used to
switch between JTAG-DP and SW-DP.
```
from the arm infocenter  [here](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0316d/Chdhfbhc.html)
```
The switcher defaults to JTAG operation on power-on reset, therefore the JTAG protocol
 can be used from reset without sending a selection sequence. Switching from one protocol 
to the other can only occur when the selected interface is in its reset state. JTAG must be 
in its Test-Logic-Reset (TLR) state and SWD must be in line-reset.
```
To this particular micro-controller, the info from arm infocenter applicable or not , I'm not sure.

* Does jtagulator supports cJTAG specification or just JTAG
* Can I use jtagulator in anyway to identify the debug pins, in this case

Thanks,
Abhijith

joegrand, 2017-05-26:
Hi Abhijith-

I'm not familiar enough with cJTAG (apparently also known as Compact JTAG, https://www.jtag.com/en/content/ieee-11497) to give you a solid answer. It looks like cJTAG has compatibility with standard IEEE 1149.1 (JTAG), but it's not clear how much. The JTAGulator currently requires all 4/5 JTAG pins and uses IDCODE and BYPASS modes for pinout identification/testing. If cJTAG doesn't support those, then it won't work. SWD is already in the TODO list (https://github.com/grandideastudio/jtagulator/blob/master/TODO.markdown) and I'll add cJTAG, as well.

Joe

hei5enberg, 2017-05-27:
Hi Joe 

Thanks for a quick reply. I've found out the SWD pins after some trial and error. (I would say its boring and time consuming :( ). 

cJTAG is becoming popular nowadays. Even the new Texas CC radio chip-sets are coming only with cJTAG (not even SWD) [link1](http://www.ti.com/general/docs/datasheetdiagram.tsp?genericPartNumber=CC2640&diagramId=SWRS176B)
Actually, it is not that new (compared to original JTAG yes it is) [link2](http://blogs.ip-extreme.com/2010/02/not-your-fathers-jtag.html)      [more info](https://www.gigaic.com/tech-talks/2009/03/24/Silvaco-Adding-Physical-Test-and-Debug-Access-to-Chips-with-a-Compact-JTAG-Core)

It's great that both these standards are on your TODO list!

Abhijith

closed 2017-05-27 by hei5enberg

cz172638, 2018-02-13:
just for record - cjtag has escape sequence to switch to 'classical' 4 pin jtag. it is used in openocd to escape from cjtag to jtag for some ti targets

hei5enbrg, 2018-04-04:
Thank you for the info. 
Like you said 4 pin cJTAG is backward compatible. But the 2 wire standard is not :\ It works on a star like topology.
Anyway it turned out the stm chip was using normal JTAG and SWD multiplexed on same pins (they call it SWJ debug port) :). There's an internal latch to recognize the debug adapter type and turn on/off the required port.

joegrand, 2020-10-27:
SWD detection is now supported as of [v1.7](https://github.com/grandideastudio/jtagulator/blob/master/CHANGES.markdown#17) 

It doesn't seem like there's been enough adoption of cJTAG to add support for it. It was hailed as the more compact successor to standard 4-wire JTAG 1149.1, but is rarely used except for some TI [CCxxxx](https://www.ti.com/tool/LAUNCHXL-CC1310) devices. There may be others, but not many. Those that have all 4 pins available for backwards compatibility should be detectable by the JTAGulator. But, those just using the 2-wire advanced version (Class T4 and above) will not. 

I'm happy to accept a PR if someone deems this feature necessary :)