grandideastudio/jtagulator #14: BYPASS scan complete! Nothing found

maxy1, 2017-06-18:
Hi, I am trying IRD STI7111 CPU It detect few pins with id scan but when I do a Bypass scan it finds nothing. What could be the issue ?

                                  UU  LLL
 JJJ  TTTTTTT AAAAA  GGGGGGGGGGG   UUUU LLL   AAAAA TTTTTTTT OOOOOOO  RRRRRRRRR
 JJJJ TTTTTTT AAAAAA GGGGGGG       UUUU LLL  AAAAAA TTTTTTTT OOOOOOO  RRRRRRRR
 JJJJ  TTTT  AAAAAAA GGG      UUU  UUUU LLL  AAA AAA   TTT  OOOO OOO  RRR RRR
 JJJJ  TTTT  AAA AAA GGG  GGG UUUU UUUU LLL AAA  AAA   TTT  OOO  OOO  RRRRRRR
 JJJJ  TTTT  AAA  AA GGGGGGGGG UUUUUUUU LLLLLLLL AAAA  TTT OOOOOOOOO  RRR RRR
  JJJ  TTTT AAA   AA GGGGGGGGG UUUUUUUU LLLLLLLLL AAA  TTT OOOOOOOOO  RRR RRR
  JJJ  TT                  GGG             AAA                         RR RRR
 JJJ                        GG             AA                              RRR
JJJ                          G             A                                 RR

           Welcome to JTAGulator. Press 'H' for available commands.

:v
Current target I/O voltage: Undefined
Enter new target I/O voltage (1.2 - 3.3, 0 for off): 3.3
New target I/O voltage set: 3.3
Ensure VADJ is NOT connected to target!
:i
Enter number of channels to use (3 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 210
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort....
TDI: N/A
TDO: 1
TCK: 5
TMS: 3
TRST#: 0
.
IDCODE scan complete!
:b
Enter number of channels to use (4 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 840
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort.......................................................................................
BYPASS scan complete!
:

maxy1, 2017-06-18:
Below you can find the full log.  Sometimes when I do a bypass test it returns the results within few seconds and it keeps on checking after that it gives bypass completed message with no results. 

Enter number of channels to use (4 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 840
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort.......................................................................................
BYPASS scan complete!
:b
Enter number of channels to use (4 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 840
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort....
TDI: 0
TDO: 1
TCK: 3
TMS: 5
TRST#: 2
Number of devices detected: 1
...................................................................................
BYPASS scan complete!
:i
Enter number of channels to use (3 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 210
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort.....
IDCODE scan complete!
:i
Enter number of channels to use (3 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 210
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort....
TDI: N/A
TDO: 1
TCK: 3
TMS: 5
TRST#: 2
.
IDCODE scan complete!
:i
Enter number of channels to use (3 - 24):  7
Ensure connections are on CH6..CH0.
Possible permutations: 210
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort....
TDI: N/A
TDO: 1
TCK: 3
TMS: 5
TRST#: 2
.
IDCODE scan complete!
:b
Enter number of channels to use (4 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 840
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort....
TDI: 0
TDO: 1
TCK: 3
TMS: 5
TRST#: 2
Number of devices detected: 1
...................................................................................
BYPASS scan complete!
:i
[... 561 lines cut ...]
Current target I/O voltage: Undefined
Enter new target I/O voltage (1.2 - 3.3, 0 for off): 3.3
New target I/O voltage set: 3.3
Ensure VADJ is NOT connected to target!
:i
Enter number of channels to use (3 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 210
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort....
TDI: N/A
TDO: 1
TCK: 5
TMS: 3
TRST#: 0
.
IDCODE scan complete!
:b
Enter number of channels to use (4 - 24): 7
Ensure connections are on CH6..CH0.
Possible permutations: 840
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort.......................................................................................
BYPASS scan complete!
:

joegrand, 2017-08-14:
Hi Maxy- 

Intermittent results could be caused by lots of factors. My first thought is there might be non-JTAG-related pins connected from the target to the JTAGulator that's causing odd responses/behavior (e.g., system reset). I haven't specifically worked with Hitachi parts before, but maybe the JTAGulator scanning process (enumerating all possible pins) is inadvertently doing something weird with HUDI? 

I noticed that when the BYPASS and IDCODE scans returned a result, it was correct (IDCODE = 0x2D43B041). That's tells me the detection behavior is working (at least when the system is in the proper state).

If you're still working on this, please try again using the latest release version of code (1.4). Also, if you feel like sending photos of your setup and any documentation/datasheets you have for your particular target, I can try to dig deeper to find the cause.

Joe

maxy1, 2017-08-14:
Hi, Thank you for your reply. Yes I checked with the version 1.4 version
and results are the same. I have the photos and datasheet. Can you send me
your email id ?

joegrand, 2017-08-14:
Sure! You can contact me at www.grandideastudio.com/contact/

closed 2017-09-01 by joegrand