grandideastudio/jtagulator #15: Feature request: SWD & cJTAG detection

ikarus23, 2017-11-30:
Hi! First of all: Thanks for this awesome tool!
I was wondering if it is possible to implement a SWD and cJTAG detection/mapping feature?

joegrand, 2017-12-02:
Glad you like the tool! Yes, it's certainly possible to implement SWD and cJTAG. It's on the TO DO list (https://github.com/grandideastudio/jtagulator/blob/master/TODO.markdown), but I can't promise if/when I'll be able to get to it :)

closed 2017-12-02 by joegrand

ikarus23, 2017-12-04:
Oh, sorry for not checking the TODO list first.
...And sorry for not checking the closed issues.

No worries, I know how it is. A lot of things would be easier if the day had 48 hours. ;)

joegrand, 2020-09-18:
SWD added in v1.7 (https://github.com/grandideastudio/jtagulator/releases/tag/1.7)

See PR #30 for limitations particularly related to the JTAGulator hardware front-end and incompatibility with many SWD-based systems due to their pull resistors.