grandideastudio/jtagulator #18: Valid Baud Rates "Out of range"

ahp95, 2018-05-21:
When doing a UART pass-through with the latest firmware, the jtagulator interface will only accept values of 300, 600, 1200, and 2400 bps. Anything else is rejected with the out of range message. Shouldn't the jtagulator support much high baud rates?

joegrand, 2018-05-21:
Yes, UART passthrough should work with all values listed in the BaudRate table at the end of JTAGulator.spin (300, 600, 1200, 1800, 2400, 3600, 4800, 7200, 9600, 14400, 19200, 28800, 31250 {MIDI}, 38400, 57600, 76800, 115200, 153600, 230400, 250000 {DMX}, 307200). 

Can you send me a screenshot/log of the response you're getting and also try the previous release of firmware to see if the problem exists there?

ahp95, 2018-05-21:
Before upgrading to the latest firmware, this problem did not exist. Here is the console log:

```
UART> p
UART pin naming is from the target's perspective.
Enter X to disable either pin, if desired.
Enter TXD pin [10]:
Enter RXD pin [8]:
Enter baud rate [300]: 9600
Value out of range!

UART >
```

joegrand, 2018-05-21:
Interesting. I must have broken something during my recent update of the UART passthrough mode. I'll see if I can replicate the issue and fix ASAP. Thanks!

ahp95, 2018-05-25:
Any luck?

joegrand, 2018-05-25:
I’d recommend just using the previous release version until I get this fixed. I’ll fix it, but can’t commit to a particular time.

ahp95, 2018-05-25:
That's weird - I'm running through PuTTY.

joegrand, 2018-05-25:
Works for me on PuTTY, too. 

Could your terminal be sending some extra hidden characters? Can you send a picture of the front/back of the JTAGulator?

joegrand, 2018-05-25:
Can you confirm how you're loading the new version? From source or pre-compiled .eeprom file?
https://github.com/grandideastudio/jtagulator/releases/tag/1.5

joegrand, 2018-05-25:
To correct an earlier comment (also edited above): Passthrough should work with all values from the minimum value listed in the table (300) to the maximum (307200). The UART cog will configure its timing for any value within that range - it doesn't have to be a specific value from within the table. So, the fact that you're seeing any baud rate within 300 to 2400 makes sense. The problem is you're being limited to 2400 when it should allow you much higher. 

The allowable baud rate detection is done in the Set_UART function:

```
if (xbaud < BaudRate[0]) or (xbaud > BaudRate[(constant(BaudRateEnd - BaudRate) >> 2) - 1])  ' If entered value is out of range, abort
```

Try to replace BaudRate[(constant(BaudRateEnd - BaudRate) >> 2) - 1] with some larger value (9600?) to see what happens. It could be some range calculation error. Without being able to replicate the issue on my side, it's going to be hard to fix without your help, since we don't know if it's a universal problem or limited to your setup.

ahp95, 2018-05-25:
Regarding firmware update method - flashed the latest firmware .spin file to the jtagulator through the propeller IDE.

I'll try reflashing the newest and some older firmware versions, and I'll give the code update a shot if I still have issues with the newest. I'll get back to you on Tuesday.

joegrand, 2018-05-31:
Great! Thanks for letting me know.

closed 2018-05-31 by joegrand