grandideastudio/jtagulator #19: Unable to find JTAG pins

jgeorge1, 2018-05-30:
I have followed the videos and have tried to get the JTAGulator working on multiple boards without any success.   Using the IDCODE scan command, it says it's JTAGulating, and then IDCODE scan is complete.  However, there are no pin numbers listed.  Firmware version is 1.2.2.

jgeorge1, 2018-05-31:
I've also upgraded the firmware to 1.5.  Still no success.  Does the JTAGulator work only on certain processors?  I've tried this on two boards, one with an ARM and the other with a TI processor.

joegrand, 2018-05-31:
The JTAGulator is not architecture specific and should be able to detect JTAG interfaces on any targets, as long as they are fully JTAG 1149.1 compliant and do not have any hardware/interfacing implementation that may affect results.

No or intermittent results could be caused by lots of factors. My first thought is there might be non-JTAG-related pins connected from the target to the JTAGulator (e.g., system reset). Ensure that GND is properly connected to the target and that there is no connection to VADJ. It could perhaps also be connectivity issues, strength of pull resistors on the target (1K ohm or less), an interface unsupported by the JTAGulator, or a known bug in JTAGulator that appears with certain targets (https://github.com/grandideastudio/jtagulator/issues/3). 

In order for me to try and help further, please provide more details about your setup (high resolution photos, information about your targets, MCU part numbers, and any screen captures). Can you also try a BYPASS Scan and let me know the results? Do you have any targets you could try where you already know the pinout?

Feel free to contact me offline (http://www.grandideastudio.com/contact/) if you don't want to share publicly.

joegrand, 2018-06-05:
Handled offline.

closed 2018-06-05 by joegrand