grandideastudio/jtagulator #20: Firmware failing to write using PropellerIDE macOS

0xff7, 2018-06-10:
Having trouble trying to upload latest firmware to my JTAGulator. I'm using latest version of PropellerIDE on macOS high sierra. After opening the JTAGulator.spin file and clicking "write" everything seems to build fine but ends on "download failed". I'm using firmware version 1.5, and have also tried 1.4. Any ideas?

joegrand, 2018-06-10:
That would be a question for support@parallax.com or https://github.com/parallaxinc/PropellerIDE. Loading firmware isn't a JTAGulator issue, per se, and sounds like it's a problem with the development environment. All JTAGulator development has been done with Propeller Tool (https://www.parallax.com/downloads/propeller-tool-software-windows-spin-assembly) and I'd highly recommend using that. Regardless of what tool you use, remember to include the entire source repo and not just the top JTAGulator.spin file.

closed 2018-06-10 by joegrand

joegrand, 2018-06-10:
FWIW, Propeller Tool is the original and most stable development environment, albeit older and requiring Windows only. Here's a direct link to video demonstrating the firmware update process:

http://www.youtube.com/watch?v=xlXwy-weG1M

joegrand, 2018-06-10:
Also, #18 had a similar problem with Propeller IDE (firmware not loading properly).

0xff7, 2018-06-26:
So just an update. While I did end up loading the new firmware in a VM with the propeller tool like You suggested I've now managed to get the same propeller tool working natively on macOS with wine+wine bottler. I haven't tried flashing a new firmware yet as I'm already up to do date but everything appears to be working just fine. Anyways for anyone else that doesn't want to use a VM this is worth checking out.

joegrand, 2018-06-26:
Looks like a promising alternative. Thanks for the tip!