grandideastudio/jtagulator #23: Perhaps UART Buffering Problem ?

shuber1, 2018-07-23:
Hi
I have a UART connection to a device which has a dump command writing memory content in hexdump style to the console. When I trigger this command to dump a longer range I have strange interrupts and line breaks in the output:

> 0x8102A9D0: 00 00 00 00 00 00 00 00 81 02 A9 C8 81 02 A9 D8 ................
> 0x8102A9E0: 00 00 00 00 00 00 00 00 FE FF FF FF 00 00 00 00 ................
> 0x8102A9F0: 00 00 00 00 00 00 00 00 81 02 A8102AA60: 00 00 00 00 00 00 00 00 FE 7F FF FF 00 00 00 00 ................
> 0x8102AA70: 00 00 00 00 00 00 00 00 81 02 AA 78 81 02 AA 78 ...........x...x
> 0x8102AA80: 00 00 00 00 00 00 00 00 FF FF EF FF 00 00 00 00 ................
> 0x8102AA90: 00 00 00 00 00 00 00 00 81 02 AA 98 81 02 A2 98 ................

or here:

> 0x810660B0: 83 05 43 A0 00 00 07 9F 81 06 60 88 81 04 58 38 ..C..........X8
> 0x810660C0: 00 00 08 24 00 00 00 02 FF 7F FF FF 83 47 FE 3C ...$.........G.<
> 0x810660D0: 83 40 2D 6... ..."..a...`X
> 0x81066140: 00 00 00 2C 00 00 00 01 FF FF FF FF 00 00 00 00 ...,............
> 0x81066150: 83 47 DA 20 00 00 00 23 81 04 61 38 81 06 61 78 .G. ...#..a8..ax

I assume it is some kind of buffering problem, because this break/ disturbance happens ever 2864 byte ?
When I use a FT232R breakout board I don't have the problem with the interruption but then I have wrong values in the output ?
My jtagulator firmware is version 1.5 llatest from release.
As console tool I used minicom, but had the same problem with screen. Perhaps some has an idea.

joegrand, 2018-07-23:
If you're seeing a similar (but not exact) problem when using a standalone FT232R breakout board, it sounds like it's not specifically a JTAGulator problem. I'm wondering if your device is the one outputting an incorrect byte every 2864 bytes, which causes either the JTAGulator or breakout board to behave strangely and handle that one byte in a different way. The JTAGulator does not use interrupts, so it may be the buffer is full and needs a moment to catch up, but I think that's unlikely. You could verify by looking directly at your target output with an oscilloscope to see what that character is ever 2864 bytes.

joegrand, 2018-07-30:
The buffer for the UART passthrough mode is 256 bytes. Upon that "glitch," are you missing 256 bytes before the dump continues? That might help to show if the problem is with the JTAGulator (perhaps the RX buffer becoming full or that errant character causing an issue) or if it's caused by the target itself (particularly because you see a problem using the FT232R also). Without being able to narrow down and replicate the issue on my side, it's unlikely I'll fix this, but would be nice to know if the cause is "mine" or "yours."

shuber1, 2018-08-02:
Sorry for late response. I don't miss the 256 byte, only 4 bytes sometimes or a linebreak (see excerpts above). I tried also now my bus pirate and with the pirate I don't have this problem. There are no "glitches". 
I am sorry that I can not help more but I am not a hardware guy, I am more in software.

joegrand, 2018-08-07:
Thanks for the detail. Since you're not missing 256 bytes at a time, it doesn't sound like a JTAGulator buffering problem. My only guess is your target is sending a non-printable ASCII character every 2864 bytes that's causing the issue. Since both the FT232R and the JTAGulator (which also uses an FT232R) will just receive and pass-through the data, the terminal program may not know how to handle that particular character (it would be nice to know what it actually is sending, but I understand you can't do that). I'm assuming the reason you don't see it with the Bus Pirate is that the Bus Pirate could be discarding bytes that aren't printable ASCII, thus preventing that "glitch" byte from ever reaching the terminal program. I'm going to close the issue, but if you end up discovering anything else that may point more towards a JTAGulator firmware bug, feel free to continue to comment!

closed 2018-08-07 by joegrand