grandideastudio/jtagulator #24: No target device(s) found!

twisted89, 2018-09-23:
I sent you an email earlier this week but not sure if you've received it so posting here too.

I'm encountering some odd results when targeting an Intel Puma 6 board, specifically a large amount of devices are shown and bypass isn't working.

Output is here https://pastebin.com/XG3VeYSZ

Let me know if you'd like me to resend all the details.

joegrand, 2018-09-23:
Can you try a BYPASS Scan with all pins and let me know the results?

joegrand, 2018-09-23:
Also, please review the closed issues and see my comments there regarding possibilities that can cause things to go wrong and/or not work. Particularly #6 and #19. Let me know if you've been able to rule out those things. 

BYPASS Scan is a much more reliable way of detection, yet much slower. If you don't see any results using that command, it's likely some of the reasons in my previous issues above are a factor. The IDCODE Scan is simply trying to shift data out on the DR upon TAP reset and if it sees something on the current pin permutation that matches a possible valid Device ID, it lists the result. A possible valid Device ID means either the LSB is 1 and the result is not all FF. That leaves lots of false positives, especially for systems that don't like being probed or perhaps aren't JTAG. 

What's the full part number for the part?

twisted89, 2018-09-23:
Thanks for the quick response, I'll wire up the other side and let you know, I'm pretty sure there isn't an issue with GND and VADJ isn't connected but I'll double check and go through the other issues you mentioned. 
Router is a Arris-TG2492, you can see a breakdown here - https://www.mobile-computer-repairs.co.uk/blog/topic/29/routers/Arris-TG2492

The exact CPU is a DHCE2652 S R278 of which I can't find any information or datasheet.

joegrand, 2018-09-23:
Sounds great! Let me know your results.

twisted89, 2018-09-25:
Before rewiring the pads I thought I'd check exactly where both sides are going. Having traced a few of them it seems that the bottom row is GND, the left side goes back to the CPU and the right side goes off to an unknown chip ZL88105 (Can't find much but I think this is some sort of wireless network chip). 
Both sides of the port are also connected to each other with resistors, I'm pretty new to the hardware side of things but I'm doubting this is even a JTAG port? 
I've simply gone with the details provided in that teardown I linked on the assumption they had identified it with some additional information but surely this indicates it's probably some bus line to the wireless interface and not a JTAG?

joegrand, 2018-09-25:
JTAG uses 4 required pins (TCK, TMS, TDI, TDO) and one optional pin (/TRST). We sometimes see additional pins on the same connector used for JTAG (e.g., /SRST for system reset, RTCK for adaptive clock, etc.). There are lots of commonly used (but not required nor a standard) interfaces used (http://www.jtagtest.com/pinouts/), many of which have one row of the double row header connected to GND. JTAG should have visible pull-up resistors connected to TMS, TDI, and TDO - perhaps those are the resistors you're seeing?

It's possible that some of those pins are the JTAG interface, but the JTAGulator can't detect it (maybe the port is disabled physically or in firmware/software), and/or some of the other lines are causing the funny results you're seeing and/or it's not a JTAG port at all. The JTAGulator is an assistive tool, meaning it should help you narrow down the interface even if there are false positives, but it's not fool-proof (yet).

joegrand, 2018-09-25:
The signals you see going off to the ZL88105 could be some other debug interface, as well, for that particular chip that the JTAGulator doesn't support (e.g., non-JTAG). Or, it could just be some signals used for diagnostics/testing that are active, which the JTAGulator is treating as false positives (e.g, it thinks it's seeing a response, but it's actually just seeing unrelated activity on those lines). You can try to not connect the obviously non-JTAG ones (the ones going to the ZL88105) to see if that changes the behavior.

blacklisted786, 2018-10-08:
drop me a line twisted working on the same thing

its not jatag you need to switch to UART they are uart points SERIAL

blacklisted786, 2018-10-08:
UART scan

twisted89, 2018-10-08:
@blacklisted786 If its a UART do you have a copy of the output? I'm curious considering there is already 2 other UART's on the board with Bootloader and CPU output so I'm not sure what use a 3rd would be?

blacklisted786, 2018-10-08:
this is the output

https://www.mobile-computer-repairs.co.uk/blog/topic/29/routers/Arris-TG2492

UART Intel Puma 6 DUMP 1 of 3 this is from ONE UART HEADER

UART ARRIS Dump 1 of 2 This is from 2ND UART header

you cannot dump or invoke shell as its raed only

may be if you carsh the soc while its outputting buy grounding the datapin we may crash boot loader?

but be carefull about this need pro advice

blacklisted786, 2018-10-08:
jtag is diabled in firmware

blacklisted786, 2018-10-08:
can you share the decrypted firmware

blacklisted786, 2018-10-08:
we may be able to patch out to UART?

blacklisted786, 2018-10-08:
or may intoduce buffer overflow may give us shell

blacklisted786, 2018-10-08:
[image]

blacklisted786, 2018-10-08:
nothing to loose with jatagulator

ricktendo, 2018-10-09:
> 
> I already have a fully decrypted dump of the firmware, I wanted to use the JTAG to debug memory at runtime.

How did you manage to decrypt the dump?

blacklisted786, 2018-10-09:
i would also like to know? come on twisted89 its all in the pot effort?

teach a man to fish

twisted89, 2018-10-09:
I'm working on a method that will allow remote shell access that I will release publicly, I don't plan on releasing the firmware at this time.

ricktendo, 2018-10-09:
Twisted,

Please tell us how to we can decrypt a full nand dump, I have a CM8200A that I would like to update with newer firmware and hopefully in the future a firmware that is shelled.

twisted89, 2018-10-09:
You can't decrypt a NAND dump read directly off the chip, the TG2492 has a Phison PS8211 emmc controller which de/encrypts on the fly with an embedded AES key. You have to crash the bootloader and read it out manually through software.

blacklisted786, 2018-10-13:
can you give insight on the software used?

twisted89, 2018-10-15:
Terminal and a custom python app to send commands & parse output.

blacklisted786, 2018-10-15:
i am using the ale pin to gnd? you have another sugestion

blacklisted786, 2018-10-22:
may i asked how you are crashing the boot procress, as i am incountering difficulty?

vladionescu, 2021-01-04:
I am running into this same situation: same unpopulated header for an Arris TG24xx with the Intel Puma 6.

JTAG scanning seems to be non deterministic and "finds" various pins, none of them appear to be sending a valid ID though. Looks like 3.3v high with the occasional low blip? There's a good chance this isn't JTAG or the port is disabled, but what could be going on? @joegrand does this output make you think of anything?

```
JTAG> i
Enter starting channel [6]:
Enter ending channel [15]:
Possible permutations: 720

Bring channels LOW between each permutation? [Y/n]: n
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort...
-------
No target device(s) found!
IDCODE scan complete.

JTAG> i
Enter starting channel [6]:
Enter ending channel [15]:
Possible permutations: 720

Bring channels LOW between each permutation? [y/N]:
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort...
-------
No target device(s) found!
IDCODE scan complete.

JTAG> i
Enter starting channel [6]:
Enter ending channel [15]:
Possible permutations: 720

Bring channels LOW between each permutation? [y/N]: y
Enter length of time for channels to remain LOW (in ms, 1 - 1000) [100]:
Enter length of time after channels return HIGH before proceeding (in ms, 1 - 1000) [100]:
Press spacebar to begin (any other key to abort)...
JTAGulating! Press any key to abort...
----------------
TDI: N/A
TDO: 6
TCK: 9
TMS: 7
Device ID #5: 1111 1111111111111111 11111100110 1 (0xFFFFFFCD)
Device ID #11: 1111 1111111111001001 11111111111 1 (0xFFFC9FFF)
Device ID #17: 0000 0111111111111111 11111111111 1 (0x07FFFFFF)
Device ID #24: 1110 0100111111111111 11111111111 1 (0xE4FFFFFF)
Device ID #30: 1111 1111111111001111 11111111111 1 (0xFFFCFFFF)

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
TDI: N/A
TDO: 12
TCK: 11
TMS: 9

--------
TDI: N/A
TDO: 12
TCK: 13
TMS: 9
Device ID #30: 1111 1111111111110111 11111111111 1 (0xFFFF7FFF)

---------------------------------------------------------------------------------------------------------
TDI: N/A
TDO: 14
TCK: 7
TMS: 11
Device ID #5: 1111 1000010111111111 11111111111 1 (0xF85FFFFF)
Device ID #18: 1111 1111111111111110 01001111111 1 (0xFFFFE4FF)
[... 169 lines cut ...]
Device ID #30: 1111 1111111111111111 10110011111 1 (0xFFFFFB3F)

--------------------------
TDI: N/A
TDO: 14
TCK: 15
TMS: 7
Device ID #4: 0111 1111111111111111 11111111111 1 (0x7FFFFFFF)
Device ID #11: 1111 1111111111111111 10001001111 1 (0xFFFFF89F)
Device ID #17: 1111 1111110001111111 11111111111 1 (0xFFC7FFFF)
Device ID #23: 1011 1111111111111111 11111111111 1 (0xBFFFFFFF)
Device ID #30: 1111 1111111111111111 11001001111 1 (0xFFFFFC9F)

-------------------------------------------------------------------------------
IDCODE scan complete.
```

Suspects:

TCK: 7, 8, 9, 10, 11, 13, 14, 15
TDO: 6, 8, 12, 14
TMS: 7, 8, 9, 11, 12, 13, 15
TRST: 15

Note that 1-5 are known UARTs, 6-15 is wired to that mystery header.

joegrand, 2021-01-05:
Hi Vlad- There are lots of reasons for behavior like that, see #2 and #3 of the [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions). Since the IDCODE Scan tries to read a valid IDCODE (if it exists) from the target, it's likely that whatever signals it's receiving back appear to be a valid IDCODE by meeting certain simple criteria when, in fact, they're not. 

I'd suggest trying a BYPASS Scan, which is a much more complete scan. It will try to determine all 4 required pins (plus the optional /TRST), most notably, TDI. If you don't get results with BYPASS Scan, then that indicates the JTAGulator isn't able to detect anything. If you *do* get a result, then it's very likely going to be an actual valid JTAG pinout. You can also try the new JTAG Scan command in the [latest firmware](https://github.com/grandideastudio/jtagulator/releases), which combines the IDCODE Scan and BYPASS Scan together.

vladionescu, 2021-01-05:
Thanks Joe! I've tried BYPASS and the new JTAG Scan without results, only IDCODE gets anything back. Sounds like my device doesn't have JTAG enabled.

joegrand, 2021-01-05:
Bummer. If JTAG is disabled, there are lots of reasons for that, too, some of which might be able to be bypassed. Or, perhaps there's a signal compatibility issue between the JTAGulator and target, so you could check the signal levels with an oscilloscope to make sure they're correct. If not, you might need to do the modifications discussed on the [Wiki](https://github.com/grandideastudio/jtagulator/wiki/Hardware-Modifications) (when I'm done writing them).