grandideastudio/jtagulator #25: Question about post-firmware outcome with JTAGulator

CyberTecky, 2018-11-18:
FW EEPROM 1.6 Loaded with an odd output effect in the Parallax Simple Terminal where the JTAGulator banner seems condensed and compacted.  Not sure if this is a result on my end?   In addition, just need to verify that available the commands are all listed properly - They seem different that the tutorial you did a few years back:

See snapshots of banner and command output (Attch)
[image]
[image]

joegrand, 2018-11-18:
It's possible that you just have a different font selected, which is causing the condensed view. You should be able to get rid of the extra linefeed between each line by unchecking the Line Feed option under Preferences. See attached images of the settings I use in Parallax Serial Terminal.

Regarding the commands, what you're seeing is the latest. The videos/tutorials used an older version of firmware and I've revised the menu system since then. Eventually, I'll make an updated video and documentation explaining functionality of the new commands. 

Enjoy!

Joe

[image]

[image]

joegrand, 2018-11-18:
It looks like the condensed view is caused by some change in font or a line length/wrap setting. Perhaps it has to do with the default system font within Windows that Parallax Serial Terminal is using. It's purely aesthetic. You should be able to get rid of the extra linefeed between each line by unchecking the Line Feed option under Preferences. See attached images of the settings I use in Parallax Serial Terminal.

Regarding the commands, what you're seeing is the latest. The videos/tutorials used an older version of firmware and I've revised the menu system since then. Eventually, I'll make an updated video and documentation explaining functionality of the new commands.

Enjoy!

Joe

[image]

[image]

CyberTecky, 2018-11-18:
Well Done Joe!

I made those changes that you suggested and everything seems to line up now.  Looks like I'm ready to begin this new adventure working with the JTAGulator!

CJ

closed 2018-11-18 by CyberTecky