grandideastudio/jtagulator #27: IDcode / Bypass scan : No results ( RPI3 )

CSamyR, 2019-03-27:
Hello, 

I am currently working on the JTagulator with a RaspberryPi 3 Model B.

I am a student in france and one of my project is the following : 
I did put Uboot bootloader with Verified Boot on my raspberrypi3 with a Kernel and my goal is to try to debug it. Just to see how it works.  

I wired my raspberry Pi 3 Jtag pins to the channels of the JTAGulator. In the RPI documentation there say that there is ALT4 mode and ALT5 mode for JTAG wiring. 

[image]

I tried both. None works.

So USB for my computer to the JTAGulator. Power imput for the raspberrypi. 

I ran putty > terminal : serial | speed : 115200 like you did on your introduction and demonstration video. ( I'm on Linux so serial line = /dev/ttyUSB0 )

I did put : 
enable_gpio_jtag = 1 in my config.txt file in the SD card of the RPI3 boot partition. 

However when i run the IDcode scan or the Bypass Scan, it says that its jutagulating and theres no devices founds.
I read all of the issues and try different way of solving my problem but nothing worked. ( Issue #3 and issue #1 were close issues to mine )

Also sorry for my English. It may not be perfect, working on it !
Thank you very much for your work

closed 2019-03-27 by CSamyR

joegrand, 2019-03-27:
I see you’ve closed the issue. Did you solve the problem? If so, please share your results.

Thanks!

Joe

joegrand, 2020-11-07:
I've confirmed that the JTAGulator does behave properly with a Raspberry Pi Zero W (BCM2835) after the JTAG pins are properly enabled. This should be consistent for other versions of Raspberry Pi. 

```Add "enable_jtag_gpio=1" to config.txt (enables ALT4 alternative peripheral function of GPIO pins)```

Alternatively, see [this guide](https://sysprogs.com/VisualKernel/tutorials/raspberry/jtagsetup/) to enable the JTAG functionality manually from a user-land program. 
 
For testing purposes, the pinout is as follows:

```
Pin #	Function    RPi Header
GPIO22  ARM_TRST    15 *
GPIO23  ARM_RTCK    16
GPIO24  ARM_TDO     18 *
GPIO25  ARM_TCK     22 *
GPIO26  ARM_TDI     37 *
GPIO27  ARM_TMS     13 *
N/A     GND         6, 9, 14, 20, 25, 30, 34, 39 (choose one) *

* = Required
```

TAPADYUTIBARAL, 2023-03-07:
Can you tell how do you solve the problem, I am also facing the same issue but not able to find out solution. IDCODE scan works for me but Bypass not working. Please Help!