grandideastudio/jtagulator #29: JTAGulator Stopped Responding

lucky-aptiv, 2019-05-16:
The JTAGUlator would not work with PropellerIDE from the Ubuntu Store. 
So I installed SimpleIDE, and with some probing, it worked. 
Then, via Propeller, I tried to update using the spin files, because the eeprom wasn't recognized by Propellar. 
Now, the hardware isn't responding, and the status LED isn't working. 
The only file that sort of ran was JDCogSerial.spin. I get a Download Failed! Message for every other spin. And SimpleIDE won't let me open Spin files for some reason, even though it has them in the documentation. 
The board is still showing up in terminal. It mounts. It identifies correctly. It has the correct driver, but now it won't do anything.

joegrand, 2019-05-16:
The code was developed using Propeller Tool for Windows. It sounds like the code didn't get loaded properly with SimpleIDE (a completely separate toolchain that we haven't verified). Please watch [this video](https://www.youtube.com/watch?v=xlXwy-weG1M) on how to properly load firmware using Propeller Tool.

Another option is instead of loading the code from source, use the .EEPROM file in the GitHub repo. This is pre-compiled. 

When the code is properly loaded, the status LED will turn on (orange) and it will appear as a virtual COM port that you can connect to with a terminal program.

lucky-aptiv, 2019-05-16:
Spinning up a Windows VM right now, will report back.

lucky-aptiv, 2019-05-16:
The device doesn't show up at all in windows. At least when connected to Linux, I can see that it's present. USB drivers were installed. Propeller was installed. Nada. 

[47957.636776] usb 1-5: new full-speed USB device number 14 using xhci_hcd
[47957.790399] usb 1-5: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[47957.790400] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[47957.790401] usb 1-5: Product: FT232R USB UART
[47957.790402] usb 1-5: Manufacturer: FTDI
[47957.790403] usb 1-5: SerialNumber: A906PQDY
[47957.793442] ftdi_sio 1-5:1.0: FTDI USB Serial Device converter detected
[47957.793461] usb 1-5: Detected FT232RL
[47957.793762] usb 1-5: FTDI USB Serial Device converter now attached to ttyUSB0

Is what shows up in Linux.

lucky-aptiv, 2019-05-16:
Tested on native install of Windows, and VirtualBox image, which doesn't show it as a USB device for me to enable to pass through to Windows.

joegrand, 2019-05-16:
Probably an FTDI driver problem with VirtualBox. I'd highly recommend just using a stock Windows install (no VM) to make sure you can get up and running. Seems like you're having the same problem as #20. 

You can also contact support@parallax.com and request a replacement if you think it's a JTAGulator hardware issue (I don't, since you're able to see it in Linux). They can help you get to the point of seeing a command prompt in a terminal program.

joegrand, 2019-05-16:
When you first received the unit, did you verify that you could communicate with it via terminal program? Before trying to update the firmware? Can you send attach a picture (front and back) of your board to this issue?

joegrand, 2019-05-16:
Also, did you try the Propeller Loader tool within Linux to load the pre-compiled EEPROM? It's mentioned in the video here: https://youtu.be/xlXwy-weG1M?t=100

Since you can see the device in Linux, you should try that if you can't find a proper Windows machine.

joegrand, 2019-05-16:
Basically, I think you're seeing one (or possibly both) problems: 

1) The firmware is incorrectly loaded onto the JTAGulator. This won't affect USB detection from the host (you can see it within Linux), but will prevent the tool from working. If you don't see the LED indicator on (orange), this indicates firmware isn't loaded and/or the device has not enumerated properly via USB.

2) Your host is not detecting the FTDI USB-to-Serial IC on the JTAGulator properly. This is purely a USB driver problem and has nothing to do with the JTAGulator per se, unless there's a hardware issue with the FT232 circuitry. If you purchased the device from Parallax, they are all tested during manufacture, so this is unlikely.

Once you solve #2 and are able to load new firmware, you'll then solve #1.

joegrand, 2019-06-09:
No response, so I'm closing the issue.

closed 2019-06-09 by joegrand