grandideastudio/jtagulator #31: OE Pin not driven high?

TheFuzzyGiggler, 2019-05-23:
Hi,

Got my Jtagulator all soldered up today and ran through the test procedures and everything looked fine. But I noticed that when setting the GPIO pins to high the voltage would consistantly be around half of the VADJ. I went to the schematic and grabbed the multimeter and started testing the expected voltages and I found the propeller chip isn't driving the OE (TXSOE) pin high so the level shifter is in 3-state mode.

It recognizes pins being pulled low if I use the "R" command on the GPIO screen and testing with the multimeter it is indeed pulled to ground. But with a VADJ of 3.3V (And verified on other pins) the voltage at the channels, on both sides of the resistor arrays and at the output of the level shifters is around 1ish volt.

I'm using firmware v1.5 and uploaded the eeprom image only. All the other chips seem to be working as expected, and the Propeller chip works in all other regards except for the OE behavior. Only other odd behavior is my LED isn't working, even though the voltage is right, but I chalked that up to a defective part.

Thanks.

TheFuzzyGiggler, 2019-05-23:
Was just looking at the spin code. It looks like the level Shifters are only enabled once it begins searching for pins. Is this correct?

adamgreen, 2019-05-23:
> It looks like the level Shifters are only enabled once it begins searching for pins. Is this correct?

I believe that that is correct. Wherever you see ```u.TXSEnable```. It is then [disabled again before each iteration of the Main loop](https://github.com/grandideastudio/jtagulator/blob/465b21961c2c83b31e88507b6e0af535eace4e1b/JTAGulator.spin#L124).

TheFuzzyGiggler, 2019-05-23:
Thanks, glad it wasn't something wrong on my end. Since you have to set the VADJ to use the GPIO function the natural assumption is that the pin would be at that VADJ voltage. I'll check the voltage on the pins when it's actively searching just to be thorough but that answers that.

Thanks again.

closed 2019-05-23 by TheFuzzyGiggler