grandideastudio/jtagulator #34: UART Search String Limited to 4 Hex Bytes

digitalentropy, 2020-04-21:
In working with a device that needs a 14 byte binary string, I ran into an issue with the JTAGULATOR being limited to 4 bytes for the initialization string.

On some devices that interact with other hardware and not a human user with a keyboard, this limitation is even more likely to come up.

joegrand, 2020-04-21:
The UART_Scan should support an input string of 16 bytes, but that maximum length might be for ASCII characters and not binary (which are defined with \x). 

Can you provide a log/screenshot showing the limitation?

digitalentropy, 2020-04-21:
```
                                    UU  LLL                                     
 JJJ  TTTTTTT AAAAA  GGGGGGGGGGG   UUUU LLL   AAAAA TTTTTTTT OOOOOOO  RRRRRRRRR 
 JJJJ TTTTTTT AAAAAA GGGGGGG       UUUU LLL  AAAAAA TTTTTTTT OOOOOOO  RRRRRRRR  
 JJJJ  TTTT  AAAAAAA GGG      UUU  UUUU LLL  AAA AAA   TTT  OOOO OOO  RRR RRR   
 JJJJ  TTTT  AAA AAA GGG  GGG UUUU UUUU LLL AAA  AAA   TTT  OOO  OOO  RRRRRRR   
 JJJJ  TTTT  AAA  AA GGGGGGGGG UUUUUUUU LLLLLLLL AAAA  TTT OOOOOOOOO  RRR RRR   
  JJJ  TTTT AAA   AA GGGGGGGGG UUUUUUUU LLLLLLLLL AAA  TTT OOOOOOOOO  RRR RRR   
  JJJ  TT                  GGG             AAA                         RR RRR   
 JJJ                        GG             AA                              RRR  
JJJ                          G             A                                 RR 
                                                                                
                                                                                
           Welcome to JTAGulator. Press 'H' for available commands.             
         Warning: Use of this tool may affect target system behavior!           
                                                                                
> U                                                                             
                                                                                
UART> V                                                                         
Current target I/O voltage: Undefined                                           
Enter new target I/O voltage (1.2 - 3.3, 0 for off): 3.3                        
New target I/O voltage set: 3.3                                                 
Ensure VADJ is NOT connected to target!                                         
                                                                                
UART> U                                                                         
UART pin naming is from the target's perspective.                               
Enter text string to output (prefix with \x for hex) [CR]: 001122334455         
Enter starting channel [0]:                                                     
Enter ending channel [0]:                                                       
Value out of range!                                                             
                                                                                
UART> U                                                                         
UART pin naming is from the target's perspective.                               
Enter text string to output (prefix with \x for hex) [001122334455]: \x001122334
4                                                                               
Value out of range!                                                             
                                                                                
UART>                                                                           
```

I can ENTER up to 12 ASCII or 10 hex bytes.  Entering more than 4 hex bytes results in "Value out of range!" error.  I think 16 hex bytes would more than cover most applications.

joegrand, 2020-04-21:
OK. There seems to be a bug in how I process the typed user data in ASCII v. hex. I might be truncating the hex prematurely based on my currently defined input buffer size. 

Will investigate shortly!

adamgreen, 2020-04-21:
https://github.com/grandideastudio/jtagulator/blob/master/JTAGulator.spin#L1160

Hex data appears to be placed in 32-bit xval variable so can't be more than 4-bytes in length.

joegrand, 2020-04-21:
You're correct. I don't remember why I did it that way - it could have been laziness/ease of coding. At quick glance, I should be able to push the hex values into the uSTR string instead of xval and then push things out the UART.

joegrand, 2020-04-23:
Fixed with latest commit 3ac9180.

This will allow up to 16 ASCII characters (e.g., ABCDEFGHIJKLMNOP) and append a CR to the end of the string OR 16 hexadecimal characters (e.g., \x00112233445566778899aabbccddeeff). There's error checking to prevent invalid characters from being used. 

I've also adjusted the UART_Scan command to properly display the previously entered string (if it exists) if it was in hexadecimal. 

I'll close the issue, but can re-open it if you find any problems!

closed 2020-04-23 by joegrand

joegrand, 2020-04-23:
[image]