grandideastudio/jtagulator #39: jtagulator firmware doesn't go beyond 1.4

iTestAndroid, 2020-10-12:
I tried with Windows AND with Linux using PropellerIDE. I can only compile and flash and get green light on board up to firmware 1.4 release. Anything above it, after flashing firmware, board is stuck at "orange light" state. Any ideas?

joegrand, 2020-10-12:
This was noted in the [CHANGES](https://github.com/grandideastudio/jtagulator/blob/master/CHANGES.markdown) for version 1.4:

> General: Wait until the user presses a key on power-up/reset before sending the JTAGulator header and command prompt.

The LED will stay YELLOW until a user presses a key. Then, the JTAGulator will send the header and command prompt. I implemented that to make sure the header was seen by all users on power-up (otherwise it was easily and often missed.

closed 2020-10-12 by joegrand

joegrand, 2020-11-07:
I've clarified this behavior in the [Quick Start](https://github.com/grandideastudio/jtagulator/wiki/Quick-Start) section of the [Wiki](https://github.com/grandideastudio/jtagulator/wiki) to hopefully make it easier for people to find in the future.