grandideastudio/jtagulator #42: vcc pin identification

yum7777, 2021-04-21:
Hi.

As I understand, the ground should be identified and then connected. (Even to remove unnecessary scans)

Need to identify the vcc pin?

If I connect the vcc pin to the channel, does the problem occur?

How can usually identify the vcc pin?

Thanks.

yum7777, 2021-04-21:
And, Is it okay to be connected to the VTRef pin?

joegrand, 2021-04-25:
Identifying and connecting GND is required, as it provides the reference point for 0V and a return path for current. You'll need to identify GND when connecting any target to any piece of equipment (not just the JTAGulator) unless you're dealing with differential signals.

The VCC pin is *not* required. The JTAGulator has its own power supply source and you should be powering your target separately, as well. The JTAGulator has input protection circuitry (see slides 21 and 22 of [this presentation](http://www.grandideastudio.com/wp-content/uploads/jtagulator_slides.pdf)) to protect itself from over-voltage on its channels. IIRC, it will clamp voltages up to 15VDC without risk of any issue. 

VTref is commonly used by a JTAG adapter to detect the voltage of the target and adjust itself accordingly. The JTAGulator doesn't have that capability, so it's not required. If you do connect VTref (or VCC) to any JTAGulator channels, that's fine and they'll just read as idle, logic level high pins. 

The downside of the input protection circuitry is that it can sometimes interfere with certain targets and won't detect them properly ala #3. This is an engineering trade-off and I chose to err on the side of caution, because I can't predict what users will connect to. You can choose to bypass the front end circuitry by [manually modifying the board](https://github.com/grandideastudio/jtagulator/wiki/Hardware-Modifications). 

If you want to identify the target's system voltage (VCC), you can find part numbers of ICs on the board, look through the data sheet for its VCC pins, and measure them on the target. Or, if you've found a likely JTAG/debug connector or test points, some of those pins will probably be pulled to a logic level high, so you can measure those and know what value to set the JTAGulator's VADJ pins.

closed 2021-04-25 by joegrand