grandideastudio/jtagulator #46: Cant load new Firmware on JTAGulator

whoot, 2021-06-04:
Hello,

I jus twanted to update the firmware of my JTAGulator to the newest version (1.11).
However, when I try to update the firmware with Propeller, I get the following error message:
[image]

[image]

Same error when I try to use the .spin file.

Its not a Propeller issue, because everything works fine when I use the 1.10 firmware!

joegrand, 2021-06-04:
The "communication stalled" error is a known issue [here](https://github.com/parallaxinc/Propeller-Tool/issues/124) and [here](https://forums.parallax.com/discussion/173090/error-communication-stalled-due-to-system-problem) with Propeller Tool 2.x and has been fixed as of [version 2.5.2](https://www.parallax.com/package/propeller-tool-software-for-windows-spin-assembly-2/). 

The problem does not exist on [Parallax Propeller Tool version 1.3.2](http://www.grandideastudio.com/wp-content/uploads/P8X32A-Setup-Propeller-Tool-v1.3.2.zip), which is [recommended for use with the JTAGulator](https://github.com/grandideastudio/jtagulator#firmware) as it's a robust and stable version.

Can you confirm which version of Propeller Tool you are using?

whoot, 2021-06-07:
Works as expected with Propeller 1.3.2
Thanks a lot!

closed 2021-06-07 by whoot

santiac89, 2023-06-05:
Sorry to open this again, I'm facing the same issue but cannot find a download link for Propeller Tool 1.3.2 anymore. I've tried version 2.5.2 and still having the same issue, that is why I'm trying to install 1.3.2 version. 

Thanks in advance

joegrand, 2023-06-05:
The file has been removed from the official Parallax site, but I have a mirrored copy on my site here:

https://www.grandideastudio.com/wp-content/uploads/P8X32A-Setup-Propeller-Tool-v1.3.2.zip