grandideastudio/jtagulator #47: Jtagulator not working - reset button heats up

security-learner, 2021-06-14:
Hi, I am using Jtagulator via putty on windows. While scanning a device, I erroneously entered 12 channels; To stop the scan, I pushed the reset button.
somehow the device stopped working. Windows sees the device, but I cannot get putty session ( it does not respond)
Status LED does not lit up or blink
I realized that the reset button is getting pretty hot.

Any idea on how to resolve it?

adamgreen, 2021-06-15:
The reset button getting hot seems weird to me. Even if it went bad and got stuck down which would hold the main microcontroller (Propeller) in reset, it should only have about 0.66mA (3.3v/5kohm) going through it. 

It would seem like that should only happen if the RESn pin of the Propeller was shorted to Vdd and then the reset button got stuck in the closed state when you pressed it.

joegrand, 2021-06-15:
This is not a failure mode that I've seen before. It shouldn't matter that you pressed the reset button to abort the scan instead of hitting any key. The reset button won't just get hot on its own. The only connection that could maybe cause it is Q1 ([schematic](http://www.grandideastudio.com/wp-content/uploads/jtagulator_schematic.pdf)), but that would surprising unless maybe there was an assembly issue or part failure.

Please post all detail as noted in [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions) item 8. The more you can provide, the easier it will be to debug your issue. And, let me know where you obtained the unit.

security-learner, 2021-06-15:
The board was working fine for several months on various devices. This time, I had following new things as compared to previous setups:
1) I already had some wires connected on the first header (only wires in air), so I used the second header for this test
2) I was testing a board with ATMega2560 on it
3) I physically traced the jtag pins from controller. However, the GND pin was not traceable. So I used a multimeter to test short-circuit from the known gnd pin on the controller to a pin on available header on the board.
4) Coming over to the PC, the board was visible and i could serially connect to it via putty
5) I was not sure about the voltage, so I tried from lower values (probably 1.5 or 1.8v not recalling)
6) It run the scan but did not identify any jtag ports. Thinking about the voltage problem, I raised it to upto 3.3v
7) As I was using 2nd header, scan was taking too much time. So I reset the board, and changed the pins to header 1
8) Once i try to reconnect the board via USB, PC could still detect the device, but i wont get putty session.
9) I tried to reset the board, but this time it was very hot (like very very hot) and the LED is also not blinking.
May be the LED has gone faulty

St66666, 2021-06-15:
I see a potential problem in your #3 If you were using the diode check beeper to find the ground, my fluke is beeping from 100 ohms down. Some low voltage power supplies if measured with an ohm meter could be under that 100 ohm limit. Are you sure what you have as a ground is really ground?

security-learner, 2021-06-15:
May be. I am not sure if it was true gnd. I have reassembled the device, will check again in few days. But even if it is not, the jtagulator should not have identified the pins; can we expect a damage in such a case?

joegrand, 2021-06-15:
This sounds like a bizarre hardware issue. The circuitry is robust and misconnection seems unlikely to cause a failure. Where did you obtain the JTAGulator? Please send high-resolution photos of the unit and your wiring configuration to the target. Also, run through the test procedure and let me know if/where it fails. 

Main link: [JTAGulator Project Page](http://www.grandideastudio.com/portfolio/jtagulator/)
Direct links: [Test procedure](http://www.grandideastudio.com/wp-content/uploads/jtagulator_testproc.pdf) and accompanying [video](https://www.youtube.com/watch?v=rFy3ZFVTLcQ)

security-learner, 2021-06-16:
I have removed it from the target. But wiring was pretty standard; Vadj not connected; Gnd to target GND (it may not be as discussed above); other JTAG pins to Channels
[image]
[image]
[image]

joegrand, 2021-06-16:
Thanks for the pictures. Since the board was originally working and then it wasn't, seems more likely that it was related to misconnection. It's still very strange behavior. Please follow the test procedure and let me know the results. Do you still get the heating issue when connecting to a known-good system (one you know works from before)?

St66666, 2021-06-16:
There is no way that a switch is heating up if it is used as it was designed here. The only way I can see this happening is if the switch connects a power supply to ground or two different power supplies together. (I can only guess what happened here)
The surge current may weld the contacts together and the switch may stay in shorted position and heat up until the power supply is disconnected or switch removed.
This can be tested with a multimeter and if it shorted - remove it.

Somehow you may have a power supply voltage on the reset pin. Maybe short between pin 7 and 8 of the U2?

security-learner, 2021-06-16:
@joegrand : Heating issue is still there when no target is connected at all.
@St66666 : I checked the reset button. There are 4 pins; 2  and 2 are shorted in pairs. But when pressed, all 4 shorted (seems good )
se pic below. I dont understand what to do with U2 
[image]

St66666, 2021-06-16:
That looks normal. Are you sure it is the switch what is heating up?
U2 is the parallax chip.
Pin 7 is reset, pin8 is 3.3V power supply. A short between the two would put 3.3V power supply on the switch.

adamgreen, 2021-06-16:
If the ground plane was getting hot, it could conduct heat into the reset switch. Are any of the other chips around the reset switch also hot? The voltage regulator (U6) is the larger 4pin device to the right of the Propeller IC and the USB load protection IC is just below the reset switch in your photos. Maybe one of those is the actual source of the heat? If so, they may have failed from a previous short or maybe there is still a short somewhere?

joegrand, 2021-06-28:
@security-learner Do you have an update on this issue?

joegrand, 2021-07-05:
No response, so issue is being closed.

closed 2021-07-05 by joegrand