grandideastudio/jtagulator #48: PulseView cannot Scan the Device

Droid-MAX, 2021-08-05:
After switching to logic analyzer mode with jtagulator using firmware version 1.11, PulseView cannot scan the device.

joegrand, 2021-08-05:
You'll need to provide way more information than that. 

Please see the [Wiki](https://github.com/grandideastudio/jtagulator/wiki/Logic-Analyzer) page for the OpenOCD mode and ensure you aren't facing a known issue. Also, check the official sigrok project [bug list](https://sigrok.org/bugzilla/). If you believe you are seeing a new issue not covered in any of the above, follow the recommendations in the [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions) item 8 to give more details about what you're seeing.

Droid-MAX, 2021-08-05:
I already follow the wiki to set, but the pulseview show nothing when I click "Scan for devices using the driver above" button.

Droid-MAX, 2021-08-05:
But after downgrading to version 1.10 firmware, can use the logic analyzer mode normally.