grandideastudio/jtagulator #50: Is there a Better USB Connector I Can Use?

Eddiiie, 2022-01-04:
Hi,
I am looking to build my own Jtagulator.      One my first board attempt the USB connector gave me the hardest time attaching to the PCB (among other ICs).   The holes did not seem to line up, the feet on the connectors had to be bent to make contact with the PCB, not really sure they ever really mated...

Now I ordered new batch of boards and a solder stencil!   I will get it this time.   But, what went wrong with the USB connector?

The BOM called for:   UX60-MB-5S8    (as you already know)

I put that into Mouser.com's search box and got the only one they had in stock...  
https://www.mouser.com/c/connectors/usb-connectors/?q=UX60-MB-5S8

SO I am guessing this is the wrong one.

Is this the wrong one?   Is there a better one?   Substitute?

joegrand, 2022-01-04:
If the components weren't aligning with the footprints, it's likely a board fabrication/scaling issue on your end. I'd check your Gerber plots or discuss with the manufacturer. The JTAGulator board files haven't changed in nearly 10 years and, as you can see from the photo, all the components properly line up :)

[image]

The USB connector you mention is the correct one. It's end-of-life, but stock is still available.

closed 2022-01-04 by joegrand

Eddiiie, 2022-01-04:
The one Mouser has in stock is not the correct one is what the problem is and there were no pictures available of that connector on their site.
I found out today that Digikey has 53,000 in stock.
Also looks like I found them on Amazon.
Excellent.