grandideastudio/jtagulator #51: setting up jtagulator for jtag

dutycyc, 2022-01-13:
hi my name is Aaron and I am having a problem with connecting to my jtagulator because it cannot find the serial port. I have specified that it was com port 3 as you can see

touchdown@ubuntu:~$ screen /dev/com3 115200
[screen is terminating]

I have typed

touchdown@ubuntu:~$ screen /dev/ttyusb3 115200
[screen is terminating]
 
this says cannot exec /dev/ttyusb3 file does not exist.

I ran touchdown@ubuntu:~$ strings $(which openocd) | grep pirate and got the fallowing

buspirate_interface
src/jtag/drivers/buspirate.c
Buspirate SWD mode enabled
usage: buspirate_pullup <1|0>
usage: buspirate_vreg <1|0>
Buspirate switched to %s mode
Shutting down buspirate.
Unable to restart buspirate!
buspirate_led
buspirate_vreg
buspirate_pullup
Buspirate %s Interface ready!
usage: buspirate_led <1|0>
buspirate
buspirate_adc
buspirate_speed
buspirate_mode
buspirate_port
usage: buspirate_mode <normal|open-drain>
usage: buspirate_speed <normal|fast>
Buspirate did not ack speed change
Buspirate did not reply as expected to the speed change command
Buspirate feature %s is not available in SWD mode
Buspirate was unable to set feature %s
Buspirate error. Is binary/OpenOCD support enabled?
Buspirate did not answer correctly! Do you have correct firmware?
Buspirate did not answer correctly
Buspirate did not reply as expected to the configure command
Buspirate two-wire mode configured correctly
buspirate_swd_init
buspirate_swd_switch_seq
buspirate_swd_read_reg
buspirate_swd_queue_cmd
buspirate_swd_write_reg
buspirate_swd_run_queue
buspirate_tap_execute
buspirate_end_state
buspirate_path_move
buspirate_tap_append
buspirate_execute_queue
buspirate_jtag_get_adcs
buspirate_handle_vreg_command
buspirate_handle_pullup_command
buspirate_handle_led_command
buspirate_handle_speed_command
buspirate_handle_mode_command
buspirate_reset
buspirate_swd_set_feature
buspirate_swd_set_params
buspirate_bbio_enable
buspirate_init
buspirate_jtag_reset
buspirate_jtag_set_speed
buspirate_serial_read
buspirate_print_buffer
buspirate_serial_write
buspirate_jtag_command
buspirate_quit

I am a novice and I am not sure what needs to be done to get the jtag part of jtaulator to work. I believe I need openocd/buspirate installed on Ubuntu and Ubuntu installed in a virtual machine which I do (VMware workstation 16 player). ubuntu is version 18.04.6 lts and the jtagulator is on firmware version 1.10. The computer I have runs windows 10. I did use (sudo apt-get install openocd) and it says that openocd is already the newest version (0.10.0-4).
I can get access to most of the serial protocols the jtagulator can interact with through realterm.
I would appreciate any help you could give.

joegrand, 2022-01-13:
I'm not clear if you're trying to get JTAGulator working at all or using the OpenOCD mode of the JTAGulator. 

For the former, please visit [Quick Start](https://github.com/grandideastudio/jtagulator/wiki/Quick-Start) on the Wiki. In particular, there's a link on how to use [screen](https://learn.adafruit.com/welcome-to-circuitpython/advanced-serial-console-on-mac-and-linux).

For the latter,  please visit [OpenOCD](https://github.com/grandideastudio/jtagulator/wiki/OpenOCD) on the Wiki. 

You can also find a bunch of [videos](https://www.youtube.com/playlist?list=PLsyTdiI7kVn8H848lMSKljkUwPnZfke9k) on YouTube. The [New Feature Update](https://www.youtube.com/watch?v=NWTNH1RgUZc&list=PLsyTdiI7kVn8H848lMSKljkUwPnZfke9k&index=2) might be the most relevant.

This should give you everything you need to get started. I'm closing the issue, because it's not related to firmware or functionality issues of the JTAGulator. Feel free to respond on the thread if you have other questions, though.

closed 2022-01-13 by joegrand

dutycyc, 2022-01-14:
Hi Mr.Grand, what I am trying to do is get my copy of openOCD on Ubuntu to interact with your Jtagulator and acknowledge it by turning the yellow light to red sir. I tried to type in the terminal screen in Ubuntu (openOCD -f interface/buspirate.cfg -c "transport select jtag; buspirate_port /dev/ttyusb0") which you showed in the last part of your "New Feature Update" on your Youtube video (at 22:45 into the video) to just get the Jtagulator to work by itself (no target device was connected to my Jtagulator). I know now that this could not work because of putting in (ttyusb0) but I did try com3 since this is what it reads in Realterm on windows 10, however, that com port may be reading differently when using Ubuntu in a virtual machine like VMware workstation 16 player. Putting in com3 did not work. I went to the link that says screen on your post and found  ls /dev/tty.*, typed it in the terminal in Ubuntu to see how the port might read and I got the fallowing (ls: cannot access '/dev/tty.*': No such file or directory). Your Jtagulator works just fine in Window 10 Mr.Grand but I just can't get it to work in VMware workstation 16 player/Ubuntu. 
 I would like to have the full rage of the Jtagulator abilities because I believe it can be a valuable tool for electronic troubleshooting. I think it is a bad idea for a microprocessor to be considered black box with question mark on top of it and having very little understanding of what is going on inside it sir. 
 I am sorry to bother again about this but I cannot find a lot videos on your Jtagulator so that I could take care of this myself.

joegrand, 2022-01-31:
This is a problem dealing with the serial port access/permissions of your Ubuntu installation and usage of OpenOCD, not specific JTAGulator functionality. If you're not able to open the port that the JTAGulator is connected to, you won't be able to use its standard command-based menu interface or the additional OpenOCD functionality. 

By using `ls /dev/tty*` you should see a list of ports, but that doesn't mean they're being used or that you have access to them.

Searching Google for `"ubuntu dialout serial port"` should give you resources on how to identify and grant proper permission to use the port. Since you're in a VM, it will *not* be the Windows COM port. 

When you *do* correctly connect to OpenOCD, while the JTAGulator is in the OpenOCD mode (solid red LED), then you should see some additional OpenOCD details and not an error like you have above, even if the JTAGulator isn't connected to a target.

dutycyc, 2022-02-01:
Mr.Grand (ls /dev/tty*) started to work as soon as I set myself as part of a dialout supplementary group like you indicated in your last post. Thank you for taking the time to communicate that information sir. Apparently any USB port I connect to will come back ttyUSB0 for the front of my computer. I would like to say that when I typed in the command (screen /dev/ttyUSB0 115200) it did not work at first but it turned out that the USB part needed to be capitalized for this string of text to execute. This was implemented in your "New Feature Update" on your Youtube video. I did run into one more problem when I typed in (openocd -f interface/buspirate.cfg -c "transport select jtag; buspirate_port /dev/ttyUSB0") I get back

 Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: Buspirate did not answer correctly! Do you have correct firmware? .

I did try to remedy this situation by installing the new firmware 1.11 that you have out on Github but this, as you can see, did not fix the problem. When using the Propeller tool it did not seem to transfer right but did show up as installed (file did not show EEPROM at the end of it but just said Jtagulator). I know you have a Youtube video on troubleshooting this board and did most of the stuff on it except for having the headers reading the zeros back to the microcontroller. I have yet to get female jumper wires and did not want to bridge the header pins. In the first post I did send you the results from the text  (string ($ strings $(which openocd) | grep pirate)  and got (Buspirate did not answer correctly! Do you have correct firmware?) and in it does show this as being a problem before in the 1.10 firmware version. This does not mean it is your fault but I do not know if it is my fault for not removing the device through Ubuntu (or Window) sir. I did start up Ubuntu while the Jtagulator was plugged in and left it in when shutting Ubuntu down by shutting down the virtual machine. I am not sure if I just need to replace the propeller microcontroller chip and install new firmware or not. I am sure you are busy sir but I would appreciate it if you could provide some insight into this problem when you have time.

joegrand, 2022-02-02:
Are you properly entering the OpenOCD mode from with JTAGulator as discussed in the [Wiki](https://github.com/grandideastudio/jtagulator/wiki/OpenOCD) before trying to launch OpenOCD?

If you believe your setup is correct, please contact [Parallax Technical Support](https://www.parallax.com/support/) directly and provide them with the information listed in [FAQ item 8](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions). Without having those details, we simply can't help you troubleshoot an issue.

dutycyc, 2022-02-03:
Hi Mr. Grand, I had the Wiki page (https://github.com/grandideastudio/jtagulator/wiki/OpenOCD) you had indicated to look at already printed out months ago and just forgot about that information sir. It described really well how to work with openOCD and I did not have any problems with getting the Jtagulator to work. There were some errors present but that was only because I did not have a target board to present to the Jtagulator (it was trying to perform the scans that it needed to do on a target board).  To be clear I consider the board in proper working order and if it is not, I will contact Propeller not you. Thank you for taking the time and patience to correspond to me. I hope that these types of interactions do not dissuade from making more electronic devices or writing more books.