grandideastudio/jtagulator #53: Firmware starting after 1.8 is not working on our jtagulator from parallax

dummys, 2022-02-23:
Hi,
I have a problem with firmware after 1.8. It is not working. There is no communication from the UART. How to debug this ?
Iam using version 1.3.2 from your website to flash it and I get no error.

joegrand, 2022-02-23:
Please see the [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions) question 6 and closed issue #40. It's likely related to the EEPROM having other data in it that caused the JTAGulator to start up in an alternate mode (new features added on 1.8 and above). 

If that doesn't solve your problem, please provide more details as mentioned in the FAQ question 8.

dummys, 2022-02-23:
OMG you are right. Thanks for explanation.

closed 2022-02-23 by dummys