grandideastudio/jtagulator #54: Scans aborting by default without waiting for the spacebar or any other key

ramyapandian1987, 2022-03-16:
Kindly help us to resolve the issue. Details are as follows - 

JTAGulator firmware version – 1.10
Description of the target hardware – Linux Raspberrypi 5.10.63-v7l+ armv7l GNU/LINUX
Connections/Wiring - 
[image]
[image]
Below is the IDCODE scan output for jtag option -
[image]
Below is the BYPASS scan output for jtag option -
[image]
•	Operating system details – macOS Monterey Version 12.2.1
•	Version numbers of any accompanying software – CoolTerm version 1.9.1
•	What steps you have already taken to troubleshoot/debug the issue
Updated settings on Raspberry Pi 4 (Target) – as mentioned in [http://www.grandideastudio.com/wp-content/uploads/jtagulator_rpi_demo_notes.pdf](https://nam12.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.grandideastudio.com%2Fwp-content%2Fuploads%2Fjtagulator_rpi_demo_notes.pdf&data=04%7C01%7Caaligombe%40towson.edu%7C855fb9a0b9e148c0978f08da06e01555%7Ccbf9739249f649dda8a619f710efcc35%7C0%7C0%7C637829857781002714%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=z65WEPuMvmwh0%2F%2F7kpW8JjMaVmXZq%2BwVsU2WV%2Fq96fU%3D&reserved=0)
 
1.	enable_jtag_gpio = 1
2.	enable_uart = 1
3.	Removed quiet from cmdline.txt
4. gpio=22-27=np

joegrand, 2022-03-16:
Thanks for the detailed report.

It's likely that the terminal program is sending both a CR and a LF, which is causing the JTAGulator to interpret it as two separate key presses. Please check your CoolTerm settings as shown below - you'll want to ensure that it's *only* sending a CR.

[image]

ramyapandian1987, 2022-03-16:
Thank you very much for the quick response! Really appreciate it.

It is working fine now. Have a great day!

ramyapandian1987, 2022-04-05:
Hello Sir,

I am trying to dump the memory of my RaspberryPi 4 through Jtagulator and OpenOCD. However, I am facing the following issue - 

[image]

Attached is the raspberrypi 4 config file that I use - 
[raspi4_config.txt](https://github.com/grandideastudio/jtagulator/files/8420718/raspi4_config.txt)

I am reopening this issue because the details of my hardwares remain the same as mentioned above.

Kindly help us resolve this issue.

Thanks!

joegrand, 2022-04-05:
This is a different issue that has to do with your host configuration, not with the JTAGulator functionality. Please see the [OpenOCD section of the Wiki](https://github.com/grandideastudio/jtagulator/wiki/OpenOCD) and closed issue #51. We aren't able to help with individual implementations like this.

closed 2022-04-05 by joegrand

joegrand, 2022-04-05:
A video demonstration of the OpenOCD functionality can be found [here](https://www.youtube.com/watch?v=NWTNH1RgUZc&t=1331s).