grandideastudio/jtagulator #55: Help

frank8383, 2022-06-24:
Good morning everyone. I have just started using this wonderful tool, and in order not to make mistakes I started with an interface with known pinouts. the strange thing is that the tool works once (finds all pins) and if I start another test it doesn't work anymore (the interface suddenly doesn’t find jtag pins). I have to unplug it several times before it gets back to work. being a newbie i upgraded to version 1.11 and i did all the tests illustrated in the guide with positive results. With the GPIO I see in real time the change of state when I connect the channels on the gnd. i can't understand and ask for your expert help. Thanks in advance

joegrand, 2022-06-25:
This is more likely a problem with how your target system is reacting to the JTAGulation process (e.g., as it's searching for the correct JTAG pins), which we can't control, or if the JTAGulator is holding other pins in a state that causes the target to misbehave. The fact that you're able to properly detect the pinout shows that the JTAGulator is working as expected.

frank8383, 2022-06-25:
thank you for the quick reply. I will try to identify the pinouts with the hope of succeeding.

closed 2022-07-03 by joegrand