grandideastudio/jtagulator #56: Not detecting any devices

nataneliezer, 2022-11-02:
We have recently started using the JTAGULATOR to try to detect both UART and JTAG interfaces. We set up gnd pins to gnd on the JTAGULATOR and with both devices powered on it doesn't seem to ever detect anything. We are on version 1.11 and have tried connecting via screen command on kali and putty on windows. We only get output while the device is starting up or in use. This has also been an issue on multiple devices.

joegrand, 2022-11-02:
We'll need way more information to be able to try and help you troubleshoot. Please review the entire [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions) and, if all that fails, provide the information as requested in FAQ #8.

nataneliezer, 2022-11-02:
- We are running version 1.11
- Our target hardware is a WinkHub 2 and we are using the UART interface. It is a smart home hub.
- We've tried using multiple devices, different pinouts, different voltages, different firmware versions, different software interfaces such as a kali VM and trying to putty in, and the propeller tool.
- Find attached images of the
- Current wiring
[image]
-  JTAGulator output,
[image]

-  and OS information
[image]

joegrand, 2022-11-02:
The UART detection will only work on actively transmitting UART interfaces. Have you confirmed that the Wink UART interface is indeed sending data? You can try the UART Passthrough command with the JTAGulator connected to communicate directly with the Wink.

You mention "We only get output while the device is starting up or in use."  Can you clarify what you mean? I'm not clear if you're using the JTAGulator properly. 

Have you tried JTAG/SWD detection, as well? I don't see connectors soldered to those footprints.

nataneliezer, 2022-11-02:
We have confirmed that the Wink UART interface is sending data.
 What we mean by only getting output while the device is starting up or in use is that we were trying to used the JTAGulator on a PCB from a mini Tetris game and when we started it up or were playing a game, we were getting results from the JTAGulator. 
We have tried to solder to the JTAG/SWD pads on a different Winkhub 2 PCB with no results. We also used a VR table to connect to the JTAG/SWD with no results either which is why you didn't see connectors soldered to the footprints in the picture.

joegrand, 2022-11-02:
> We have confirmed that the Wink UART interface is sending data. 

Since you know the UART pins of the Wink, have you tried the JTAGulator's UART passthrough mode to verify proper communication?

Since you've tried with multiple JTAGulators, it's unlikely a hardware issue, but you can try to run through the test procedure to make sure: [Test procedure](http://www.grandideastudio.com/wp-content/uploads/jtagulator_testproc.pdf) and accompanying [video](https://www.youtube.com/watch?v=rFy3ZFVTLcQ)

Were these purchased directly from Parallax? Can you provide photos of the backsides?

> What we mean by only getting output while the device is starting up or in use is that we were trying to used the JTAGulator on a PCB from a mini Tetris game and when we started it up or were playing a game, we were getting results from the JTAGulator.

OK, so are you saying the JTAGulator is working as expected, but just not with the Wink? Can you provide screenshots of the output?

joegrand, 2022-11-08:
Closing due to no response.