grandideastudio/jtagulator #58: Inconsistent results in identifying UART or JTAG

killab443, 2023-02-24:
Hello all. I recently purchased a flashing device for my honda vehicle. The device allows me to add custom fuel tunes to the oem ECU in the honda vehicle through the OBD port. The device is a Hondata flashpro. The only issue is, I bought it second hand and it is currently VIN locked to another vehicle (seller screwed me). The unlock can only be performed with the original vehicle, and the original seller has blocked me. Hondata charges 500$ to unlock the device and remove the vin (crazy). This is to prevent people from using the device on multiple vehicles. They want you to spend 700$ on a new one... So, since I am out 500$ minimum because the seller screwed me over, I would like to see if I can read some firmware off of the device. The device plugs in USB into my PC and then it has an OBD pigtail that plugs into the car. While running the Hondata software, the unlock takes seconds when on the correct vehicle( I have witnessed this on a friends flashpro on his own vehicle), which leads me to believe it is a relatively simple reflash that occurs.

I recently purchased a jtaguator (FW 1.11), I have started to tinker with the onboard 8 pin header that is available. The only issue is, I can't tell if it is Jtag or UART. I have done manual tests(attached) with a multimeter which make it seem that there is no jtag device avail. In jtag scan, bypass scan, and IDscan it does not find any devices.( see attached). When doing a UART scan, I get mixed results. If I ignore non printable characters I get mixed results. Once it shows Pin 7 TXD, Pin 4 RXD, another time Pin 8 TXD, Pin 4 RXD. When attempting to use either of these pins for bypass, the device does not respond. I am confident in the ground I am using because I have probed the board at several locations and it seems correct. I checked voltage throughout and that also seems 3.3v is correct. I am confused as to where to go from here. If anyone can help me interface with this I will greatly appreciate it! I tried to be as detailed as possible with photos and info/logs attached

A few more things to note:
There is a carrot on the first pin that is on the targer device. I assume that this is pin one. So I am hooked up CH1-Ch8 on the jtagulator with respect to that. Pin 1 also seems to be a ground.
The device has a lpc2368fbd100 MCU, datasheet(attached) shows that it has UART0, UART1, UART2, and UART3, as well as JTAG. 
It also a NAND Chip 29f2g08abaea datasheet(attached) has helped me identify the VCC

[UART-IGNORE.txt](https://github.com/grandideastudio/jtagulator/files/10826818/UART-IGNORE.txt)
[UART-IGNORE2.txt](https://github.com/grandideastudio/jtagulator/files/10826819/UART-IGNORE2.txt)
[UART-WITH ALL.txt](https://github.com/grandideastudio/jtagulator/files/10826820/UART-WITH.ALL.txt)
[ByPASS SCAN.txt](https://github.com/grandideastudio/jtagulator/files/10826821/ByPASS.SCAN.txt)
[image]
[image]
[IDCODE SCAN.txt](https://github.com/grandideastudio/jtagulator/files/10826825/IDCODE.SCAN.txt)
[JTAG SCAN.txt](https://github.com/grandideastudio/jtagulator/files/10826826/JTAG.SCAN.txt)
[LPC2364_65_66_67_68.pdf](https://github.com/grandideastudio/jtagulator/files/10826827/LPC2364_65_66_67_68.pdf)
[image]
[micron_technology_mict-s-a0001032698-1-1759218.pdf](https://github.com/grandideastudio/jtagulator/files/10826828/micron_technology_mict-s-a0001032698-1-1759218.pdf)

killab443, 2023-02-24:
Small update here. I unplugged the usb to the device and rebooted it during a passthrough and received the following attached output. not really sure what any of these characters mean
[Uploading UART-REBOOT.txt…]()

wyattearp, 2023-02-26:
Looks like the baudrate on the UART is messed up

joegrand, 2023-02-26:
The JTAGulator itself appears to be functioning as expected. Could be that the UART speed is correct, but the data sent is not all ASCII or that the UART speed changes during boot. The LPC2368 that's on your target board supports JTAG, but is likely disabled via Code Readout Protection ([data sheet](https://www.nxp.com/docs/en/data-sheet/LPC2364_65_66_67_68.pdf) pg. 36).

I'm closing the issue here because it is related to usage/target, not a bug in the JTAGulator. @killab443 cross-posted this to my [Discord server](https://discord.com/invite/wud8KnF2Gm), so you can follow the conversation there if interested.

closed 2023-02-26 by joegrand