grandideastudio/jtagulator #61: is there any way to debug why uart passthrough is not outputting when doing tx rx connected together test(for echo test)?

kocoman1, 2023-04-15:
is there any way to debug why uart passthrough is not outputting when doing tx rx connected together test(for echo test)? thx

joegrand, 2023-04-16:
Have you enabled Local Echo when you started the passthrough command?

https://github.com/grandideastudio/jtagulator/blob/master/JTAGulator.spin#L1996

joegrand, 2023-04-16:
[image]

joegrand, 2023-04-21:
No response from OP, assuming the issue was solved.

closed 2023-04-21 by joegrand

kocoman1, 2023-11-15:
sorry I didn't have time to look at it till now..

Its giving me a▒aaa▒aa
its receiving ▒, should i need some logic probe to debug it? what chip is broken do you think? thanks

joegrand, 2023-11-17:
Are you using an officially manufactured JTAGulator or are you building your own? You can try to follow the [test procedure](http://www.grandideastudio.com/wp-content/uploads/jtagulator_testproc.pdf) to see if that narrows down the problem. Though, if you're able to properly communicate with the JTAGulator, it's likely not UART hardware related. Without knowing anything about your setup, hardware, baud rate you're testing with, etc., I'm really not able to give you any specific suggestions.