grandideastudio/jtagulator #62: N976V in EDL mode challenge

infr-automation, 2023-04-16:
Hi,

I have challenge for you https://github.com/bkerler/edl/issues/380

There is no cryptowallets on the device. For the challenge we can put a cryptowallet on a similar device and brick it the same way. 😁

joegrand, 2023-04-21:
This is a request for project help and not a JTAGulator bug. I'm closing the issue, but comments remain open.

closed 2023-04-21 by joegrand

infr-automation, 2023-04-22:
@joegrand,
A bit more serious than just one device, The new Samsungs and Qualcomms,

They've increased security, which sadly prevent root and prevent many optimizations and "Right to Repair" actions, such as, EDL unbrick, band selections as well as impose politically weird annoying regional restrictions.

This is why I am asking you as well, if you can give your opinion on the matter, or possible course of action, not just for this device, but for the bigger picture lack of root and lack of control, Please do.

Thank you.