grandideastudio/jtagulator #65: jtagulator locked

VraiHack, 2023-08-24:
Hi all,
Few days ago i got my first "Jtagulator" from Mouser Electronics , and i was doing a simple Demo to test its functionality by targeting an [Elbert V2 - Spartan 3A FPGA](https://numato.com/docs/elbert-v2-spartan-3a-fpga-development-board/).

since i already now the JTAG port of the Target (already tagged), i only used connected to the Jtagulator the 4 essential pins (TCK, TMS, TDO and TDI), and ground. And the Jtagulator was capable to identify them only for the TMS wasn't (TMS = N/A)

while i am recording the demo to my instagram page for my fans, i noticed that the LED status turned and maintain  RED, and minicom connection shows "device loked",

The weird thing is that, i still capable to see the Jtagulator in my driver list in both windows and Linux machine, but no output appear on the monitor. And the target output voltage is 3.3 on the JTAG pin (i measured and also that mentioned and the product description of the target board : https://numato.com/docs/elbert-v2-spartan-3a-fpga-development-board/)

I pressed the Reset button, problem remain always the same.

What do you suggest on me in my troubleshooting ?
Thanks

joegrand, 2023-08-24:
Please see [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions) item 6. It's possible that the JTAGulator is in the OpenOCD or Logic Analyzer modes. 

If the device isn't detected by minicom, it could also be a port/access issue. Try unplugging the JTAGulator and using it on a different machine to compare the results.

VraiHack, 2023-08-24:
Thanks Joe, we still learning from you :), by following this my problem solved: 
[image]

VraiHack, 2023-08-24:
Problem solved by following FAQ https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions

closed 2023-08-24 by VraiHack