grandideastudio/jtagulator #67: USB Connector Obsolete

miller-craig, 2023-11-21:
UX60-MB-5S8 is obsolete and no one has any in stock.  I've been searching for a replacement but appears the available have different footprints.  Any found a suitable replacement?

wesk1, 2024-04-15:
> UX60-MB-5S8 is obsolete and no one has any in stock. I've been searching for a replacement but appears the available have different footprints. Any found a suitable replacement?

i found it here, its model C
https://www.aliexpress.us/item/3256806185457786.html

nathanjackson, 2024-05-02:
YMMV on these connectors. I bought some and it’s a really tight fit. I haven’t attempted to solder them because I found some of the UX60-MB-5S8 connectors.

joegrand, 2025-06-27:
It appears that Mouser currently [has stock](https://www.mouser.com/ProductDetail/Hirose-Connector/UX60-MB-5S870?qs=QNEnbhJQKvbDWNWR7WLL%2FA%3D%3D) of the original connector. They may also have a suitable replacement like the [UX60SC-MB-5S8(80)](https://www.mouser.com/ProductDetail/Hirose-Connector/UX60SC-MB-5S880?qs=PABxe4V6HDrKXwhoQapS3A%3D%3D), but I haven't checked the footprint exactly. Of course, that AliExpress link is good, too!