grandideastudio/jtagulator #68: Subject: Unable to Detect UART Activity on ESP8266MOD Module or Raspberry PI 3

VraiHack, 2024-03-20:
Hi, I'm encountering an issue while attempting to use the JTAGulator to discover a know UART pins on an ESP8266MOD module and Raspberry PI3 ([i followed this same steps described in this tutorial](https://www.instructables.com/JTAGulating-the-Raspberry-Pi-2/)). 

Despite uploading code to enable UART communication on the ESP8266MOD module, and [enable Serial port on RPi3 the JTAGulator](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjzjtC-24GFAxVETaQEHeCRDksQFnoECA4QAQ&url=https%3A%2F%2Fwww.instructables.com%2FRaspberry-PI-3-Enable-Serial-Communications-to-Tty%2F&usg=AOvVaw0eoFguZ3pMFUjLf5rUy3Rf&opi=89978449) fails to detect any UART pins.

Picture 3:
[image]

I will appreciate your help :) to capt the issue.

**Jtagulator board information:**
#32215 Rev B
2013 by 3.0 US
SMT 117291-0135

Btw this the first time i am testing UART since i purchased last year Aug 24, 2023.

joegrand, 2024-03-20:
Have you confirmed that the Raspberry Pi is indeed sending and/or receiving UART data on those pins? 

The JTAGulator is only going to detect UART if there is data being sent from the target device at the time of JTAGulation or if the target device responds to the text string that the JTAGulator sends. If there is no activity, the JTAGulator won't detect anything. 

See these videos for more details on UART commands:

[JTAGulator: UART Discovery](https://www.youtube.com/watch?v=yYCDOWI1xfo&list=PLsyTdiI7kVn8H848lMSKljkUwPnZfke9k&index=7)
[JTAGulator: New Feature Update](https://www.youtube.com/watch?v=NWTNH1RgUZc&list=PLsyTdiI7kVn8H848lMSKljkUwPnZfke9k&index=2)

VraiHack, 2024-03-20:
Thanks, Joe. I confirmed on my oscilloscope that no data is being sent on the UART serial0 of the Pi 3. 

**Curious question please:** still i can scan during the boot process of the PI3 if the JTAGulator have a triggering option to time the scan ? as i know during the system boot process UART0 will be used by the PI or not ?
[image]

I am not sure what i am saying but if there is a trigger option on the JTAGulator to timing the the scan, i believe it will capt the UART right ?

### Any way, Problem solved according to the issue i opened:
To render this matter conducive for others attempting to replicate and establish it as a proof of concept, it's imperative to affirm that the issue does not stem from any deficiency on the JTAGulator's end.

**Step 1:** on the pi enable serial interface (from raspi-config or boot.config, instruction can be found [here](https://www.instructables.com/Raspberry-PI-3-Enable-Serial-Communications-to-Tty/)).

**Step 2:** wrote this script on the PI
`import serial
import time

# Configure serial0 port (usually UART0)
ser = serial.Serial(port='/dev/serial0', baudrate=115200, timeout=1)

try:
    while True:
        # Send a predefined text string over serial0 (UART0)
        ser.write(b'Hello from Raspberry Pi!\n')
        time.sleep(1)  # Adjust the delay as needed
        
        # Read data from serial0 port (UART0)
        data = ser.read(ser.in_waiting)  # Read all available bytes
        if data:
            print("Received:", data)
except KeyboardInterrupt:
    ser.close()
`

**Step 3:** connect and setup the JTAGUlator (once you hit the space button to start the scan, run the previous python script)
[image]

On the PI side you will capt this:
[image]

closed 2024-03-21 by VraiHack