grandideastudio/jtagulator #71: Cant load firmware 1.12 on JTAGulator

yooo1999, 2024-09-02:
I read [that](https://github.com/grandideastudio/jtagulator/issues/46) but it does not help me.

I tried to update jtagulator 1.6 to 1.12 in Windows

I tried  Propeller 1.3.2 when I load EEPROM  (c:\Users\USER\Downloads\jtagulator-1.12\JTAGulator.eeprom )I got

Propeller chip lost on COM8

joegrand, 2024-09-02:
https://forums.parallax.com/discussion/97251/hardware-communication-error-fixed
https://forums.parallax.com/discussion/170359/propeller-lost-on-com-port

You could also try using one of the alternative development environments: 
https://github.com/grandideastudio/jtagulator?tab=readme-ov-file#firmware

bigttys0, 2024-09-03:
I sincerely appreciate Joe's dedication to the Jtagulator repo. This is the only repo I subscribe to and every time I break out the Jtagulator I know Joe is always there on the other end to assist. Thanks, Joe!

yooo1999, 2024-09-03:
I have tried
[image]
Latency 16 and 12 and same error

joegrand, 2024-09-03:
They have posted the same issue on my [Discord](https://discord.com/invite/wud8KnF2Gm) and have yet to provide the information requested in [FAQ](https://github.com/grandideastudio/jtagulator/wiki/Frequently-Asked-Questions) item 10 though I've asked multiple times. It's too hard to help debug without knowing about their specific JTAGulator hardware, computer/OS environment, etc.

joegrand, 2024-10-14:
No response from OP.

closed 2024-10-14 by joegrand